victronenergy/venus #896: Fronius ESS Assistant Grid setpoint

sunnyday333, 2022-01-29:
Dear team,

Today I found out, that as soon as the sun is up and the Fronius PV inverter turns on, the CCGX can see that it has turned on (sunspec model 101 or 103, there is an operating status field). When this happens, it increases the grid setpoint. There is some logic maybe to how much, but generally, it increases the grid setpoint by about 70 watts. In other words, it instructs the Multi and Carlo Gavazzi to attempt to hold the grid import to whatever I set to my grid setpoint (on the ESS menu) plus about 70 watts. 
CCGX ignores actual PV inverter power, which leads to higher grid consumption. Is this bug or feature due to regulation?
Is it possible to implement a second grid setpoint that will be active if the PV inverter is on to minimize this behavior?

System details:
Fronius Primo 5.0-1 (fw: 0.3.26.0) connected to AC-Out of the Multiplus
Multiplus 48/5000/70-100 (fw: 2624481)
Cerbo GX (fw: 2.73)
2x SmartSolar 250/70 (fw: 1.59)
2x BMZ ESS (fw: 214.11)
DVCC is activated

Thank you.

sunnyday333, 2022-02-02:
Dear team,

To bring it closer after a detailed look. The start and end of production on the Fronius inverter is much slower than with Victron DC MPPT controllers. In order to avoid frequent surpluses left in the grid, I propose to implement the possibility to define by how many percent or watts the output from the Fronius inverter should be lower than load and to replace the missing remainder from Victron MPPT regulators. This will significantly prevent the mentioned surpluses into the grid.

As you can see in the picture:
the orange color is the actual power of Fronius
green is the current consumption
the red color is a grid

[image]

sunnyday333, 2022-02-11:
Hi,

I have noticed today, that there is an incorrect regulation of Fronius PV inverter via Modbus from Cerbo GX when the battery is charged to 100% as you can see:
[image]
There is a lot of fluctuations caused by Fronius PV inveterer because Cerbo GX often changes limit via Modbus. Dashed line is a limit from Cerbo GX.

mpvader, 2022-02-11:
Hi @sunnyday333 , we'll look into it, thank you.

sunnyday333, 2022-02-17:
Hi @mpvader ,
I want to ask if there is anything new in solving this problem?

Flohman, 2022-05-27:
I have exactly the same issue and would like to vote this issue up. ;)
This happens even during charging the battery with the multiplus 2. As far as I know this is not allowed in all countries. This is not a big deal, for me at the moment as I charge the battery via MPPT during summer. But I have to use the charger during winter time.

sunnyday333, 2022-07-19:
Hi @mpvader,
is there something new?
I want to provide more relevant graphs where it can be seen that the problem occurs when the Fronius inverter's performance is attenuated via Modbus from the Cerbo GX.
[image]
[image]

As you can see, after 9:12 the consumption of the house was higher than the actual power of Fronius. that's why the surpluses to the grid disappeared.