victronenergy/venus #897: Brick VenusOS with USB

AlbertronicDev, 2022-01-31:
Hi,

After testing some installation code, it seems like we have accidently bricked our Cerbo. I think that, to sum it up, this happens because the /data/rcS.local file contains a reboot instruction.

Using a USB drive with the Victron restore file (venus.tgz) which wipes out all the files on the /data does not help.
Is it correct that rcS.local is called rearlier than unpacking a venus-data.tgz/tar.gz file? If yes, can this be changed? It would mean that you can brick the Cerbo, even on purpose (like a virus) by making a venus-data.tar.gz with a rcS.local file which contains the reboot command.

Multiple fixes possible in case of rcS.local is called earlier than the unpacking:
Fix 1. first call unpack, then rcS.local
Fix 2. Use the button for safe booting. If the button is pressed during boot, do not call rcS.local/rc.local

mpvader, 2022-01-31:
Hi @AlbertronicDev , noted, thank you. We'll look into this.

jhofstee, 2022-05-20:
The order is the following:

/etc/rcS.d/S99custom-rc-early.sh (rcS.local)
/etc/rc5.d/S30update-data.sh (update /data)
/etc/rc5.d/S99custom-rc-late.sh (rc.local / initscript)

The rcS.local is there to hook in early in the boot process. Obviously you have to be extremely careful there, things might not be there yet: network / USB sticks / sd-cards et cetera (and hence their ability to restore things). Since the sole purpose of the rcS.local file is to run early, it should be run early. Commonly rc.local should be used instead.

closed 2022-05-20 by jhofstee