victronenergy/venus #898: Remote Console on LAN used with Safari gets slower and slower

mpvader, 2022-01-31:
Until one kills Safari and starts again.

Seen on a macbook pro, by Stefanie:

```
macOS 12.2 Monterey
Chip Apple M1 Pro
16 GB
Safari Version 15.3 (17612.4.9.1.5)
```
She got that macbook today, and updated it to macOS 12.2 today, and has the issues since today as well.

The websockify log shows that the number of connections becomes higher and higher, see log below. And slowly over time the Remote console on LAN becomes slower and shows less and less pixels. See screenshot.

At the same time, Remote Console on VRM works fine - that doesn't use websockify.

My guess is that the issue is that the code in the browser is causing issues. Perhaps this is as simple to using a newer [noVNC](https://github.com/novnc/noVNC/releases) library.

```
2022-01-31 18:14:30.258304500 *** CCGX booted (0) ***
2022-01-31 18:14:38.437248500 *** starting websockify-c ***
2022-01-31 18:14:38.445148500 Warning: 'self.pem' not found
2022-01-31 18:24:00.031054500   0: target closed connection
2022-01-31 18:24:00.031775500 Waiting for connections on 0.0.0.0:81
2022-01-31 18:24:00.031779500   0: got client connection from 192.168.178.34
2022-01-31 18:24:00.031782500   0: forking handler process
2022-01-31 18:24:00.031784500   0: using plain (not SSL) socket
2022-01-31 18:24:00.031787500   0: using protocol HyBi/IETF 6455 13
2022-01-31 18:24:00.031790500   0: connecting to: 127.0.0.1:5900
2022-01-31 18:24:00.031792500   0: handler exit
2022-01-31 18:24:26.818104500   1: client closed connection
2022-01-31 18:24:26.818819500 Waiting for connections on 0.0.0.0:81
2022-01-31 18:24:26.818823500   0: got client connection from 192.168.178.34
2022-01-31 18:24:26.818826500   0: forking handler process
2022-01-31 18:24:26.818829500   1: got client connection from 192.168.178.34
2022-01-31 18:24:26.818831500   1: forking handler process
2022-01-31 18:24:26.818834500   1: using plain (not SSL) socket
2022-01-31 18:24:26.818836500   1: using protocol HyBi/IETF 6455 13
2022-01-31 18:24:26.818839500   1: connecting to: 127.0.0.1:5900
2022-01-31 18:24:26.818841500   1: handler exit
2022-01-31 18:25:16.048860500 Waiting for connections on 0.0.0.0:81
2022-01-31 18:25:16.048865500   0: got client connection from 192.168.178.34
2022-01-31 18:25:16.048868500   0: forking handler process
2022-01-31 18:25:16.048871500   1: got client connection from 192.168.178.34
2022-01-31 18:25:16.048873500   1: forking handler process
2022-01-31 18:25:16.048876500   2: got client connection from 192.168.178.57
2022-01-31 18:25:16.048878500   2: forking handler process
2022-01-31 18:25:16.048881500   2: using plain (not SSL) socket
2022-01-31 18:25:16.048883500   2: using protocol HyBi/IETF 6455 13
2022-01-31 18:25:16.048964500   2: connecting to: 127.0.0.1:5900
2022-01-31 18:25:16.048967500   2: client sent orderly close frame
2022-01-31 18:25:16.048970500   2: handler exit
2022-01-31 18:25:19.118155500 Waiting for connections on 0.0.0.0:81
2022-01-31 18:25:19.118160500   0: got client connection from 192.168.178.34
2022-01-31 18:25:19.118163500   0: forking handler process
2022-01-31 18:25:19.118165500   1: got client connection from 192.168.178.34
2022-01-31 18:25:19.118168500   1: forking handler process
2022-01-31 18:25:19.118170500   2: got client connection from 192.168.178.57
2022-01-31 18:25:19.118173500   2: forking handler process
2022-01-31 18:25:19.118175500   3: got client connection from 192.168.178.57
2022-01-31 18:25:19.118226500   3: forking handler process
2022-01-31 18:25:19.118228500   3: using plain (not SSL) socket
2022-01-31 18:25:19.118231500   3: using protocol HyBi/IETF 6455 13
2022-01-31 18:25:19.118233500   3: connecting to: 127.0.0.1:5900
2022-01-31 18:25:19.118236500   3: client sent orderly close frame
2022-01-31 18:25:19.118238500   3: handler exit
2022-01-31 18:28:25.232524500 Waiting for connections on 0.0.0.0:81
2022-01-31 18:28:25.232529500   0: got client connection from 192.168.178.34
2022-01-31 18:28:25.232532500   0: forking handler process
2022-01-31 18:28:25.232534500   1: got client connection from 192.168.178.34
[... 729 lines cut ...]
2022-01-31 22:06:36.752282500   16: got client connection from 192.168.178.57
2022-01-31 22:06:36.752347500   16: forking handler process
2022-01-31 22:06:36.752350500   17: got client connection from 192.168.178.57
2022-01-31 22:06:36.752353500   17: forking handler process
2022-01-31 22:06:36.752355500   18: got client connection from 192.168.178.57
2022-01-31 22:06:36.752358500   18: forking handler process
2022-01-31 22:06:36.752360500   19: got client connection from 192.168.178.57
2022-01-31 22:06:36.752363500   19: forking handler process
2022-01-31 22:06:36.752386500   20: got client connection from 192.168.178.57
2022-01-31 22:06:36.752389500   20: forking handler process
2022-01-31 22:06:36.752391500   21: got client connection from 192.168.178.57
2022-01-31 22:06:36.752394500   21: forking handler process
2022-01-31 22:06:36.752397500   22: got client connection from 192.168.178.57
2022-01-31 22:06:36.752399500   22: forking handler process
2022-01-31 22:06:36.752402500   23: got client connection from 192.168.178.34
2022-01-31 22:06:36.752425500   23: forking handler process
2022-01-31 22:06:36.752428500   24: got client connection from 192.168.178.57
2022-01-31 22:06:36.752430500   24: forking handler process
2022-01-31 22:06:36.752433500   24: using plain (not SSL) socket
2022-01-31 22:06:36.752435500   24: using protocol HyBi/IETF 6455 13
2022-01-31 22:06:36.752438500   24: connecting to: 127.0.0.1:5900
2022-01-31 22:06:36.752522500   24: client sent orderly close frame
2022-01-31 22:06:36.752525500   24: handler exit
root@raspberrypi4:~# 
```

mpvader, 2022-02-01:
comment from Stefanie, @AlexImb :

> A problem this morning: I can no longer reproduce it this morning or since yesterday after I did force kill safari. The LAN remote console opens immediately and without lag. I even restarted the macbook. Even after that everything seems to be back to normal.
The only thing that remains are the entries in the websockify log. And I noticed that a ‘connection’ to the iPhone (192.168.178.34) is also being established. However, the iPhone hasn’t opened a single app, not even as a background process. I did an iPhone update to iOS 15.3 this morning. Then a fresh restart. No apps open, all Safari tabs closed. As soon as I open the remote console via macbook I see 192.168.178.34 again via ‘cat /data/log/websockify-c/* | tai64nlocal’. Maybe only advertising aka mDNS instead a real connection?

mpvader, 2022-02-01:
And more:

> This how it starts in the log. Always with the client connection to the iPhone:
2022-02-01 08:35:03.440887500 Waiting for connections on 0.0.0.0:81
2022-02-01 08:35:03.440891500   0: got client connection from 192.168.178.34
2022-02-01 08:35:03.440894500   0: forking handler process
2022-02-01 08:35:03.440897500   1: got client connection from 192.168.178.34
2022-02-01 08:35:03.440899500   1: forking handler process

> And I even might get two additonal ip addresses if I start my other macbook and iPad.

nmbath, 2022-02-01:
If you need any particular logs, I can provide from both a MacBook Pro M1, which I have been running for almost 2 months without issues, or my CerboGX or a RaspBerry Pi3 running VenusOS.

I have never seen any lag or perfoamcne issues with remote console. But do have these in my logs...

2022-02-01 07:56:34.365318500 Waiting for connections on 0.0.0.0:81
2022-02-01 07:56:34.365326500   0: got client connection from 10.100.5.176
2022-02-01 07:56:34.365329500   0: forking handler process
2022-02-01 07:56:34.365330500   1: got client connection from 10.100.5.176
2022-02-01 07:56:34.365333500   1: forking handler process
2022-02-01 07:56:34.365335500   2: got client connection from 10.100.5.176
2022-02-01 07:56:34.365337500   2: forking handler process
2022-02-01 07:56:34.365339500   3: got client connection from 10.100.5.158
2022-02-01 07:56:34.365479500   3: forking handler process
2022-02-01 07:56:34.365484500   4: got client connection from 10.100.5.176
2022-02-01 07:56:34.365487500   4: forking handler process
2022-02-01 07:56:34.365489500   5: got client connection from 10.100.5.158
2022-02-01 07:56:34.365491500   5: forking handler process
2022-02-01 07:56:34.365493500   6: got client connection from 10.100.5.158
2022-02-01 07:56:34.365495500   6: forking handler process
2022-02-01 07:56:34.365497500   7: got client connection from 10.100.5.158
2022-02-01 07:56:34.365499500   7: forking handler process
2022-02-01 07:56:34.365525500   8: got client connection from 10.100.5.158
2022-02-01 07:56:34.365527500   8: forking handler process
2022-02-01 07:56:34.365529500   9: got client connection from 10.0.10.194
2022-02-01 07:56:34.365531500   9: forking handler process
2022-02-01 07:56:34.365533500   10: got client connection from 10.0.10.194
2022-02-01 07:56:34.365536500   10: forking handler process
2022-02-01 07:56:34.365538500   11: got client connection from 10.0.10.196
2022-02-01 07:56:34.365540500   11: forking handler process
2022-02-01 07:56:34.365653500   12: got client connection from 10.0.10.194
2022-02-01 07:56:34.365657500   12: forking handler process
2022-02-01 07:56:34.365659500   13: got client connection from 10.0.10.193
2022-02-01 07:56:34.365719500   13: forking handler process
2022-02-01 07:56:34.365722500   14: got client connection from 10.0.10.194
2022-02-01 07:56:34.365724500   14: forking handler process
2022-02-01 07:56:34.365726500   15: got client connection from 10.0.10.197
2022-02-01 07:56:34.365728500   15: forking handler process
2022-02-01 07:56:34.365730500   16: got client connection from 10.0.10.197
2022-02-01 07:56:34.365732500   16: forking handler process
2022-02-01 07:56:34.365734500   17: got client connection from 10.0.10.197
2022-02-01 07:56:34.365755500   17: forking handler process
2022-02-01 07:56:34.365757500   18: got client connection from 10.0.10.197
2022-02-01 07:56:34.365760500   18: forking handler process
2022-02-01 07:56:34.365761500   19: got client connection from 10.0.10.197
2022-02-01 07:56:34.365764500   19: forking handler process
2022-02-01 07:56:34.365766500   20: got client connection from 10.0.10.194
2022-02-01 07:56:34.365768500   20: forking handler process
2022-02-01 07:56:34.365770500   21: got client connection from 10.0.10.196
2022-02-01 07:56:34.365787500   21: forking handler process
2022-02-01 07:56:34.365790500   22: got client connection from 10.0.10.196
2022-02-01 07:56:34.365792500   22: forking handler process
2022-02-01 07:56:34.365794500   23: got client connection from 10.0.10.196
2022-02-01 07:56:34.365796500   23: forking handler process
2022-02-01 07:56:34.365798500   24: got client connection from 10.0.10.194
2022-02-01 07:56:34.365800500   24: forking handler process
2022-02-01 07:56:34.365802500   25: got client connection from 10.0.10.197
2022-02-01 07:56:34.365819500   25: forking handler process
2022-02-01 07:56:34.365821500   26: got client connection from 10.0.10.194

These are my home network, which has an oVPN connection to the boat...
10.194 is a MAC Mini running MacOS 12.2
10.197 is an iPad Air 3 running IOS 15.1
10.196 is an iPhone running IOS 15.1
10.193 is a Mac Book Pro M1 running MacOS 12.2

The 5.x address is on the boat... But neither of the 5.176 or 5.158 address are on the boat at the moment....

root@Venus:~# ping 10.100.5.176
PING 10.100.5.176 (10.100.5.176): 56 data bytes
^C
--- 10.100.5.176 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss
root@Venus:~# ping 10.100.5.158
PING 10.100.5.158 (10.100.5.158): 56 data bytes
^C
--- 10.100.5.158 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss

Unfortunately I cant map these to devices due to Apple changing the Mac address on each WIFI network, also my DHCP server is showing a lease, but no device name mapped to it. I can say that it is likely the MacBook Pro and iPhone.

I hope this information might be of help, and if you need anything else please do ask, or go and probe my VenusOS.

nmbath, 2022-02-01:
My home RPI device, which has no device on at the moment has nothing in the log file..

root@raspberrypi2:~# tail -f /var/log/websockify-c/current | tai64nlocal 
2022-01-29 17:40:28.240322500 Warning: 'self.pem' not found
2022-01-29 17:42:15.978220500 *** CCGX booted (0) ***
2022-01-29 17:42:29.629678500 *** starting websockify-c ***
2022-01-29 17:42:29.639655500 Warning: 'self.pem' not found
2022-01-29 18:33:53.832749500 *** CCGX booted (0) ***
2022-01-29 18:34:05.312616500 *** starting websockify-c ***
2022-01-29 18:34:06.413835500 Warning: 'self.pem' not found
2022-01-29 19:01:06.121365500 *** CCGX booted (0) ***
2022-01-29 19:02:53.835018500 *** starting websockify-c ***
2022-01-29 19:02:53.843576500 Warning: 'self.pem' not found
^C
root@raspberrypi2:~# cat /opt/victronenergy/version 
v2.80~38
Victron Energy
20220115193611

My Mac mini has been connected to the remote console for about an hour.

AlexImb, 2024-02-02:
I'm also on M1 Mac, most of El Niño is, and we did not notice any slowness. We have some laptops in the office with VRM and Remote console open for hours at some point. 

@mpvader I suggest we close this and I'll take care of closing it on our side too.