victronenergy/venus #901: Improve automatic selection of battery monitor and BMS used for DVCC

izak, 2022-02-02:
When the battery monitor is set to "default" in Settings -> System Setup, this is the situation now:

Up to Venus 2.73 it works like this:

* The default battery monitor, if none is selected, is the service with the lowest alphabetically-sorted value
  * This loosely translates to the lowest DeviceInstance, since the DeviceInstance is traditionally allocated from the serial port name for a BMV, but can cause some types to be preferred over others.
  * It means a CAN-bus BMS is automatically selected over a BMV, since `socketcan` sorts before `tty` (good, but messy!).
* The BMS used for DVCC is the same as the battery monitor IF this battery is a BMS (publishes a charge voltage)
* If the default battery is not a BMS, the battery service that is a BMS (has a charge voltage) and with the lowest DeviceInstance is used for DVCC.
* The BMS selection happens out of sight (under water), which one is used is not communicated to the end user (bad!).
* If the end-user/installer explicitly selects a BMV as battery monitor, DVCC automatically finds a BMS to use.

In Venus 2.80 we [attempted to improve][1] BMS selection by always making it the same as the battery monitor.
* The default battery monitor, if none is explicitly selected, is the service with the lowest alphabetically-sorted value (unchanged from 2.73)
* A CAN-bus BMS therefore still takes precedence over a BMV.
* The BMS used for DVCC is assumed to always be the same as the battery monitor (whether auto- or manually selected)
* If the default battery service is not a BMS, then the system is assumed to have none (not good).
* If the end-user/installer explicitly selects a BMV as battery monitor, typically for better SOC or voltage tracking, then DVCC has no BMS to use (bad!)

Suggested improvements to auto-selection of battery monitor.
* Use the battery service with the lowest DeviceInstance that is also a BMS
  * This is **new** behaviour. It can theoretically cause some systems to switch to a different battery monitor.
* If no BMS in system, select battery service with lowest DeviceInstance
  *  This is **new**, but because the Deviceinstance is allocated from the serial terminal for BMVs, would not affect which BMS/Smartshunt is auto-selected. A mild change.

Suggested improvement for selection of DVCC BMS:
* Introduce a setting where the user can configure which battery he wants to use for DVCC. This defaults to auto-select
  * This is a **new** feature.
  * If a specific battery is selected as BMS service, use that for DVCC.
* If no explicit selection, then auto-select:
  * If system battery monitor is a BMS, use that (same as 2.80 and 2.73).
  * If battery monitor is not a BMS (which can only happen if it was manually selected), use the battery service with the lowest DeviceInstance that is also a BMS
    * this restores the selection algorithm used in 2.73 and earlier!
* Communicate the BMS selection via DBUS to the GUI (**new**)
* Add the ability to select a BMS via the GUI. (**new**)

## TODO
- [ ] systemcalc: improve system battery monitor selection
- [ ] systemcalc: add setting for selecting which BMS to use
- [ ] systemcalc: split selection of system battery monitor and DVCC-used BMS, implement BMS selection
- [ ] systemcalc: Publish available BMS information to dbus for GUI use
- [ ] systemcalc: Publish /ActiveBmsInstance so other services know which one is in use.
- [ ] systemcalc: Publish something user-friendly to show what BMS is used in the GUI (DVCC menu)
- [ ] GUI: implement BMS selection and information under DVCC menu
- [ ] hub4control: Use /ActiveBmsInstance from systemcalc to use the correct battery

[1]: https://github.com/victronenergy/venus/issues/847

izak, 2022-07-25:
Implemented in branches.

Gui branch izak/bms-selection
hub4control branch bms-selection
systemcalc branch bms-selection-2

To be merged soon in the next development cycle (after 2.90).

closed 2022-10-17 by jhofstee

jhofstee, 2022-11-14:
The summary of the changelog: For systems having multiple BMSes connected, allow selecting which one should be used for DVCC. It also allows the use of a BMV for SOC tracking, by selecting BMV as battery monitor, while still using the BMS for DVCC.

mman, 2024-01-19:
@izak Sorry to interrupt this already closed bug, but I think I may have found a bug in the implementation where controlling VE.Bus BMS v2 is not correctly displayed in a system with Victron Smart LiFePo, SmartShunt acting as BMV, and VE.Bus BMS v2 with Multiplus 12/1600 and MPPT Smart Solar. Details here: https://community.victronenergy.com/questions/256027/dvcc-with-vebus-bms-v2-cerbo-gx-mppt-smart-solar-1.html