victronenergy/venus #904: DVCC regulation

sunnyday333, 2022-02-02:
Dear team,

I found out that when DVCC is switched on, the regulation does not work exactly as intended. As soon as the battery is charged (CCL is 0A), Venus OS reduces the production from the regulators and does not take into account the actual consumption of all components (regulators, multiplus etc.. and the battery is constantly discharged. This condition continues up to SOC 96% when the battery reaches CCL at some charging level. This situation does not occur if I turn off the DVCC, then the regulation is exactly as intended.

[image]

Thank you

mpvader, 2022-02-02:
Hi @sunnyday333 this is by design. All compatible managed batteries don't work with CCL, instead they lower their charged voltage in case they have (for example) a cell on which they want to reduce the voltage.

sunnyday333, 2022-02-02:
Hi @mpvader 
Yes I understand, but CVL (Charge Voltage Limit) is constant. It does not change, so it is strange to me why this discharge occurs.

sunnyday333, 2022-02-03:
Hi @mpvader .

Today I tried the island mode and the regulation is stable in this mode. It does not even drain the battery as I wrote above

[image]
.

mpvader, 2022-02-03:
Correct. To work with our equipment, the battery must *not* set CCL=0.

If the battery sets CCL to 0, then in ESS mode the system will discharge. And in island mode there is no problem.

The solution is to change the software in the battery. It should use CVL, not CCL.

Alternatively, use one of officially compatible batteries. See this list: https://www.victronenergy.com/live/battery_compatibility:start.

I'm sorry, but we really cannot help you any further here: this CVL / CCL control is a complex topic and we don't give support just like that to anyone on this. I hope you understand.

To get more help, in case you are a sizeable battery manufacturer: please contact your Victron distributor, and ask them to contact our sales manager.

Or try this forum: https://community.victronenergy.com/spaces/31/mods.html.

sunnyday333, 2022-02-03:
Hi @mpvader

My battery is on the list of compatible batteries, so I assume that DVCC has been verified and functional when it is forced on from version 2.80. The battery is BMZ ESS Z.

Thank you.

mpvader, 2022-02-03:
Hi @sunnyday333 , that is strange. We will look at that & discuss with BMZ. Meanwhile, the best solution for your system will be to revert to v2.73 I'm afraid. I'll get back to you tomorrow; we should then also move this to a support case, rather then continue here on a github ticket.

Perhaps you can tell me the name of your system? Then I can find it in VRM, analyse it, and also find your email address and contact you from our support system, via email.

sunnyday333, 2022-02-04:
Hi @mpvader,
Thank you, I'll wait for the ticket to open. In the meantime, I downgrade to version 2.73.
I changed the name in the VRM portal to E2BSMK753DWZ for security reasons.

Well, thank you.

mpvader, 2022-02-04:
Hi, thank you - found it. You can change the name again if you want. And you'll receive an email today. I'll close this issue now.

closed 2022-02-04 by mpvader