victronenergy/venus #905: Add fahrenheit temperature unit

mpvader, 2022-02-06:
With the temperature monitoring becoming a thing now, and US sales as well, this is getting requested a lot.

Tank level units are configurable already ([see here](https://github.com/victronenergy/gui/blob/master/qml/PageTankSensor.qml#L45)), and temperatures are settable to Fahrenheit on VRM.

And rather than having a system wide unit setting configurable in every temperature sensor settings menu, I think its better to add the setting to the Display & language menu.

kwindrem, 2022-02-06:
A common temperature unit setting makes the most sense.

I've added one for my GuiMods which I'll remove when one is added to Venus.

Mine affects only the temps tiles in the flow and Mobile overviews customized as part of GuiMods. It also allows display of both Fahrenheit and centegrade to be shown at the same time. but I don't think that's completely necessary and takes space on the GUI.

jhofstee, 2022-03-22:
@jepefe  one for you, https://github.com/victronenergy/venus/issues/918

jhofstee, 2022-03-22:
@kwindrem 

Since I added this to the c++ part, which isn't publicly available, this it what it does:

/**
 * The VeQuickItem can perform some unit conversions.
 *
 * VeQuickItem {
 *   sourceUnit: Unit.Celsius // optional
 *   displayUnit: Unit.Fahrenheit
 *
 *   sourceValue: 50
 *   sourceMin: 0
 *   sourceMax: 100
 *   decimals: 1
 * }
 *
 * Will define a value between 0 and 100 °C with a value of 50°C displayed in Fahrenheit:
 *   // min -> 32 °F
 *   // max -> 212 °F
 *   // value -> 122 °F
 *   // unit -> '°F'
 *   // text -> 122.0 °F
 *
 * For a simply bound Item adding an unit should enough to support different units.
 *
 * VeQuickItem {
 *   uid: "something"
 *   displayUnit: Unit.Fahrenheit
 * }
 *
 * min / max can be assigned e.g. to limit it to the value of another
 * value.
 *
 * NOTE: the source min / max will change along! sourceMin / sourceMax
 * can also be set and min / max will change accordingly.
 *
 * NOTE: the min / max can differ slightly from what is assigned due to rounding errors.
 * In normal cases that shouldn't happen though.
 *
 * The non overwritten (as in the values defined by the item) are available as
 * defaultMin / defaultSourceMin to allow construct like:
 *
 * min: (condition ? something : defaultMin).
 * The sourceUnit is optional, a default will be used when not set.
 */

closed 2022-03-22 by jhofstee