victronenergy/venus #909: NMEA2000 CerboGX Tank140 instances configurable

DevoDave, 2022-02-20:
Raymarine cannot cope with tanks with NMEA2000 device instances greater than 5. My  setup is a CerboGX with a Tank140 and I have two tank level senders connected to the Tank140. Looking at the NMEA2000 data, the device instance is set to 10 and 11. To the best of my knowledge I cannot modify these settings via the CerboGX gui or my Raymarine setup. (I do have an Actisense NGT but again cannot modify this particular PGN)

mpvader, 2023-03-26:
Tbd during upcoming n2k/tank meeting. Pls put on the agenda @martinbosma, thanks

martinbosma, 2023-03-29:
@DevoDave We found and fixed an issue related to changing fluid level (data) instances. To my knowledge Raymarine is using the data instance in the fluid level PGN, not the device instance. 

This fix is scheduled to be included in the next VenusOS version (v3.00). The fix enables the fluid level instance to be changed using Actisense NMEA Reader. However, it look like there is also a bug in the Actisense tool. It seems to combine the fluid level instance and fluid type (both 4-bit fields) as a single 8-bit value as selector. I raised this issue at Actisense today and they will look into it and hopefully provide a fix soon.

The device instance is also configurable on the devices page, found in Settings -> Services -> VE.Can port -> Devices and then goto the device you want to change.

DevoDave, 2023-03-29:
Fantastic. When 3.00 drops I'll give it a try.

closed 2023-03-30 by mpvader

mpvader, 2023-03-30:
Hey @DevoDave , to add to above, the fix is included per beta version v3.00~29. Which I'll make available tomorrow.

Thank you for reporting! I assume that many people, whom are already in difficult waters when having to deal with NMEA2000 instances, will be very thankful for having this fixed.

DevoDave, 2023-03-30:
Sweet. I'm away from my vessel for a week but that's the first job I'll do
when I return.

Thanks for hopefully fixing this.