victronenergy/venus #912: dbus-generator: Remove distinction between generic relay-controlled and FischerPanda generators

izak, 2022-02-28:
On dbus service `com.victronenergy.generator.startstop0` the paths are different depending on whether you're using a FischerPanda generator (CAN-bus controlled), or you are controlling it using the built-in relay.

When using a FischerPanda generator, all paths are prefixed with `/FischerPanda0`, while it is prefixed with `/Generator0` in other cases.

This distinction should be removed.

TODO:
- [ ] Change dbus-generator so only one module is active at a time (FischerPanda takes priority), and remove prefix on dbus paths
- [ ] Change generator tracking in systemcalc
- [ ] Update GUI QML
- [ ] Update vrmlogger attributes
- [ ] Update modbus-tcp paths 
- [ ] It will be necessary to rebuild venus-docker; Let developers know.

kwindrem, 2022-02-28:
When this gets rolled in, please notify me as I'll need to make changes to my GeneratorConnector package due to the path changes.

kwindrem@icloud.com

izak, 2022-03-01:
modbus-tcp will be unaffected. Everything stays in the same place.

MQTT: some paths will change.

closed 2022-08-17 by mpvader