victronenergy/venus #922: vebus: Add serial number(s) to dbus and log to VRM

izak, 2022-03-28:
There is a need to be able to show the serial numbers of Multis/Quattros on VRM, on the device list.

Currently this information is not available on dbus, but since firmware 308 it is possible to obtain it. Older firmware versions return 0, which can simply be displayed as invalid on dbus.

TODO
- [ ]  Add serial number readout to vebus_interface
- [ ] Add the information to a suitable place in the GUI, possibly on the Device page as with other types
- [ ] Log Serial number to VRM

vejpasop, 2022-05-05:
Implemented in mk2-dbus v3.39

On older (<308) versions the dbus path "Devices/#/SerialNumber" remains invalid

mpvader, 2022-08-17:
Jos, Izak, can this issue be closed?

closed 2022-08-18 by izak