victronenergy/venus #930: vrmlogger: logging is done every 15 minutes even with 1 hour interval set

SturmUDrang, 2022-04-08:
I tested on a cerbo that was factory reset with 2.84 and no devices connected. I set he log interval to one hour and yet, every 15 minutes there is a connection to the VRM. The connection itself and the payload was verified by running the vrm logger in debug mode. 
Also reproduced on 2.72

SturmUDrang, 2022-04-15:
I checked a little the way it works and it seems that 

because `seconds_until_next_kwh_delta_upload()` is set to return 900 secs all the time, the `send_hourlydeltas()` makes a send every 900 secs regardless of the interval settings

my workaround was to declare ` global state` and
`
def seconds_until_next_kwh_delta_upload():
        if state.settings['interval'] <= 900:
                return 900 - (int(time.time()) % 900)
        else:
                return state.settings['interval'] - (int(time.time()) % state.settings['interval'])
`

jhofstee, 2022-05-20:
yes, what you set is the default periodic log interval, it will react on state changes + kwh changes as well. Perhaps we should document this somewhere, but it isn't a bug.

closed 2022-05-20 by jhofstee

SturmUDrang, 2022-05-20:
Hi! I know, but I still think it is a bug. Because of
seconds_until_next_kwh_delta_upload() , the system triggers a state change
every 15 minutes, so values above 15 minutes don't make sense..So anything
that you set above 15 minutes is never used because an automatic state
change every 15 minutes is triggered.

jhofstee, 2022-05-20:
they do make sense for period logs (I guess), but not for triggered logs like these.

SturmUDrang, 2022-05-20:
Yes but whatever you set. There is always a trigger every 15 mins

Az iPhone készülékemről küldve

jhofstee, 2022-05-20:
also one events as well, I am bit in the dark what you're after, you can disable vrmlogger completely if that is what you want.

SturmUDrang, 2022-05-20:
I just want it to log every hour. With this code regardless of what you set as a logging interval, an event is generated every 15 mins that triggers an upload. So if I am not mistaken, log intervals above 15 mins are pointless 

Az iPhone készülékemről küldve

jhofstee, 2022-05-20:
What you set with the log interval is the default log interval, On event basis or e.g. for the kWh counters, it will trigger a log.

mpvader, 2022-05-23:
Correct. That is for the kWh data; 9 orso values

That can’t be changed easily: also VRM database expects those per 15 minute. A higher interval can, or at least might, create issues in the graph data quality.

This is not something we can easily change.

The only thing we can do is change the documentation to make this clear.

rest of the data (not the kWh deltas) is sent at the interval selected by user and also in case a state changes.