victronenergy/venus #933: Vebus: Add support for VE.Bus BMS V2 firmware updates through VRM

vejpasop, 2022-04-14:
VE.Bus BMS V2 firmware can be updated through vbdup. We need to add the update functionality through VRM.

- [ ] Add to support to mqtt-rpc

See https://github.com/victronenergy/mqtt-rpc/tree/WIP_vebus_bms_v2_update as a starting point. With this in place VRM lists the VE.Bus BMS V2.
[image]

- [ ] Add support to VRM

1. A special connection type "vebusBms" has been defined. With this connection type the mk2vsc (remot vebus configuration) steps should be skipped. 

2. The help/warning screens should be adapted for the BMS V2 update procedure

3. The "VE.Bus BMS and Digital Multi Contol Specifics" help/warning screen for the normal Multi firmware update procedure should contain a note that is possible to update Multi firmware when a VE.Bus BMS V2 is connected. Or alternatively since VRM already knows that the a V2 BMS is connected. It could just show a similar message with info about the "Digital Multi Control" only.

[image]

- [ ] Fix mk2-dbus BMS firmware version readout to be independent from "redetect system"

Currently mk2-dbus only re-reads the BMS version number after a "redetect system". Since we want  to show the correct after firmware update version on VRM this should be fixed.

vejpasop, 2022-04-14:
Added fix to mk2-dbus v3.37

mpvader, 2023-03-26:
What still needs to be done here @vejpasop ?

vejpasop, 2023-03-31:
This issue is completed

closed 2023-03-31 by mpvader