victronenergy/venus #937: Seplos BMS 48V CAN with Waveshare RS485 CAN HAT

atze09, 2022-04-24:
Hello community,

I have a question concerning the CAN communication between my RPI Running Venus 2.85 with the Waveshare RS485/CAN HAT connected via CANbus to my Seplos BMS.

Via "/data/VeCanSetup/setup"

I installed the #9 Waveshare CAN
[image]

can0 Waveshare Hat 12 MHz crystal
can1 Waveshare Hat 12 MHz crystal

The i see the CAn Bus in the Venus software

[image]

But i get the ERROR-Passive failure.
In the Seplos Software i choose "Victron under CAN Protiocol"
I have double checked the wiring and measured the wires but still the same isse.

Hope you can help me.

Thank you!

Best Regards,

Dominik

nmbath, 2022-04-24:
I am sure that Victron will close this and direct you to community.victronenergy.com for support from within the community.

Have you looked at these links and followed these steps?

https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-image

and especially this one?

https://github.com/victronenergy/venus/wiki/RaspberryPi-CAN-Interfaces

Have you checked that the can interface is being detected and configured at the OS layer. Try using the can-utils candump command to check if there is any traffic being received.

I am assuming based on the statement about Via `"/data/VeCanSetup/setup"` that you are using one of Kevins utilities. He does monitor and reply to posts on the community forum. Also have you checked the wiki page https://github.com/kwindrem/VeCanSetup. While it is great bit of automation, when things go wrong the basics help and the links above should help.

atze09, 2022-04-24:
in parallel i opened the same at community.victronenergy.com. i also tried candump can0
  can0  305   [8]  00 00 00 00 00 00 00 00
  can0  307   [8]  12 34 56 78 56 49 43 00
  can0  351   [8]  40 02 00 00 9E 07 D0 01
  can0  355   [8]  64 00 64 00 00 00 00 00
  can0  356   [8]  DE 14 00 00 C6 00 00 00
  can0  35A   [8]  AA AA AA 02 AA AA AA 06
  can0  35E   [8]  53 48 45 6E 65 72 67 79
  can0  35F   [8]  10 00 02 07 20 01 00 00
  can0  305   [8]  00 00 00 00 00 00 00 00
  can0  307   [8]  12 34 56 78 56 49 43 00
  can0  351   [8]  40 02 00 00 9E 07 D0 01
  can0  355   [8]  64 00 64 00 00 00 00 00
  can0  356   [8]  DE 14 00 00 C6 00 00 00
  can0  35A   [8]  AA AA AA 02 AA AA AA 06
  can0  35E   [8]  53 48 45 6E 65 72 67 79
  can0  35F   [8]  10 00 02 07 20 01 00 00

But still eroor-passive in venus.

atze09, 2022-04-24:
After playing with config there is still an error passive, but the SHEnergy and battery was added to Venus. So this can be closed.

closed 2022-04-24 by atze09