victronenergy/venus #939: Unable to remove Ruuvi temperature sensor

antiguasky, 2022-04-28:
Unable to remove a Ruuvi temperature sensor from VRM. The installation is running version 2.82-large-30.  

The Ruuvi tag has been disabled on the Cerbo GX and has been removed it from the device list in VRM but it is still present in the panel.

mpvader, 2022-05-20:
Hi, I tested the same, and it works just fine.

Can you post your VRM url here? (it won't allow anyone else to see your data)

antiguasky, 2022-05-20:
Thank you for taking a look.  The VRM url to the installation is below.

https://vrm.victronenergy.com/installation/37304/dashboard <https://vrm.victronenergy.com/installation/37304/dashboard>

mpvader, 2022-05-20:
Hi, thanks. I fixed it for you. I'll close this issue.

Some notes my self and colleagues: that sensor had only been sending data before we added the /status path. And therefore it didn't show up in the device list, and as such the user can't delete it himself.

Fixed by removing those rows from the database (lastLogData). There could be a few more of such systems out there, I'll ask Alex to check the database for more and then just delete those.

antiguasky, 2022-05-20:
Thank you for the help.

closed 2022-05-20 by jhofstee