victronenergy/venus #941: AC Load energy meter not showing in dashboard

matt1309, 2022-05-07:
Setup PV and loads on AC in of multiplus. 

Energy meter on PV and Energy meter on loads. 

Venus OS doesnt then do the calculation of AC loads - PV = Grid.... Is there a way to enable this?

It works when the formula is rearranged. ie Energy meter = grid meter and PV then Grid +  PV = AC loads.

jhofstee, 2022-05-20:
Indeed, it doesn't. I doubt it could become stable or even be allowed.

closed 2022-05-20 by jhofstee

matt1309, 2022-05-20:
Apologies for my lack of understanding but why is it any less stable. Due to calculation speeds? 

ie
If a grid meter is installed
AC Loads is calculated by grid + pv

Yet there's stability concerns of using an AC load meter and grid being calculated as 
AC loads - PV = grid?