victronenergy/venus #947: modbus-tcp: Add registers for VE.Bus energy paths

izak, 2022-05-16:
We make energy paths available for many devices (pv-inverters, grid meters), but not for the Multi. Also add:

```
/Energy/AcIn1ToAcOut
/Energy/AcIn1ToInverter
/Energy/AcIn2ToAcOut
/Energy/AcIn2ToInverter
/Energy/AcOutToAcIn1
/Energy/AcOutToAcIn2
/Energy/InverterToAcIn1
/Energy/InverterToAcIn2
Energy/InverterToAcOut
/Energy/OutToInverter
```

mpvader, 2022-05-16:
(I changed my previous comment, non volatile was wrong, its stored volatile)

jhofstee, 2022-05-20:
> The reason that we don't make them available is that they are (a) stored volatile in the Multi (a power cycle will make them restart at 0) and (b) on top of that they overflow on a quite quickly, to restart at 0.

The driver handles the quick overflows, just a implementation detail, it shouldn't be visible to the outside world. (Or you mean something else then I have in mind..)

izak, 2022-05-20:
More detail on this. We have a customer who wants to replicate the energy-accounting we do in kwhdeltas in vrmlogger. His original question was for a "safe range" where he can allocate his own modbus registers, for things we don't map, such as these energy counters.

Since the vebus energy counters are really the only parts missing, it might have been a good idea to simply add them.

But I am convinced that the quick-overflow nature is sufficient reason not do this right now. So I am closing this. Already contacted the customer with the info he needs to do it himself.

izak, 2022-06-02:
I have now ascertained the facts on this matter.

1. The driver does indeed handle the wrapping of the individual counters it gets from each Multi, and hides it. The published energy counters on dbus does not wrap frequently under normal operation.
2. There are however TWO ways in which the counters will reset to Zero.
  * Turning the Multi off and back on will restart the energy counters at zero. This is not unexpected, the same is true for the Phoenix, Inverter RS and Multi RS.
  * Rebooting the GX device -- for example during a firmware update -- will restart the counters. This is different to other hardware.

On balance I am now leaning towards adding them, and documenting it well. I've since had another customer asking about these counters, and since modbus-TCP is typically used by technically-minded people, and a wrapping/zeroing counter already has to be catered for, I'm going ahead and adding them.

closed 2022-06-03 by jhofstee