victronenergy/venus #950: Charge current limit in DVCC does not support inverter and MPPT in parallel

weichenb, 2022-05-22:
I like to refer to an problem with DVCC functionality, which is not working according to [description of DVCC manual.](https://www.victronenergy.com/media/pg/CCGX/en/dvcc---distributed-voltage-and-current-control.html).

**Issue:**
In case a a charge current limit is defined in DVCC and PV power exceeds that limit, it is considered by not directing AC-OUT connected inverter power to battery anymore, but MPPT power still flows completely into battery instead of being fed in grid by Multiplus.

**Deviations from DVCC manual:**
-     8.4.1. 1) In case DVCC charge limit is smaller than CANBUS-BMS limit, GX/MPPT does not care.
-     8.4.1. 2) The manual refers, that this mechanism does not work with non-Victron MPPTs. However it unfortunately does also not work for Victron SmartSolar MPPT VE.Can 250/100 rev2 (at least in the setup described below). 
-     8.4.1. 3) The manual refers that the functionality might not work without a SmartShunt. Since DC current is exactly know by GX because of the CANBUS-BMS, there should be not need for a Shunt. When BMS reports a lower limit, MPPT power is also redirected to grid. Why not in the case of the DVCC current limit?

**Potential Solution**
Since CANBUS-BMS related charge limits are exactly considered - also for a mixed AC-Out related inverter and Victron MPPT mix - the implementation would be quite simple considering following logic
MIN(<BMS charge limit>;<DVCC charge limit>)

**Setup:**
-     MultiPlus-II 48/5000/70-48 in 3-phase config
-     Fronius Symo 6.0-3-M on AC-Out
-     SmartSolar MPPT VE.Can 250/100 rev2
-     BYD LVS battery

buddhafragt, 2023-07-09:
no solution until now? still not work.

weichenb, 2023-07-19:
No, behaviour is unfortunately unchanged.