victronenergy/venus #957: USB "Permission Denied" error - Venus OS Large on Raspberry Pi with Actisense NGT1 USB

amas008, 2022-06-08:
I have installed the Venus OS Large v2.90~10 software (with SignalK enabled) on a Raspberry Pi v4. I have both a Victron VE.Direct Cable (connected to a BMV-712) and an Actisense NGT1 connected to my Raspberry Pi via USB ports. (The NGT1 is to receive NMEA 2000 data.) 

My problem is that, in SignalK, I cannot make the NGT1 work as SignalK reports a "permission denied" error when SignalK tries to open the NGT1 USB port. (See screenshot below.)

In SignalK under "Server", I have added the NGT1 as a "Data Connection" with Data Type "NMEA2000". It is "Enabled". The NMEA 2000 Source is "Actisense NGT-1 (canboatjs)". "Serial port" is 'by id' using /dev/serial/by-id/usb-0403_NGT-1-A_1A4FA-if00-port0, baud rate 115200. "Collect Network Statistics" is on. "Use Can NAME in source data" is off.

When I add this, SignalK gives no error. When I reboot the Raspberry Pi and open the SignalK Dashboard I then get a SignalK error report about the NGT1 Data Connection: The Dashboard shows, for my NGT1 data connection, "Error: Permission denied, cannot open /dev/serial.... [same serial port ID as above]. A screenshot is below.

I know the Actisense NGT1 is successfully reading data at the 115200 baud rate as I can see the PGNs on my Windows machine, and also on a standard SignalK installation (i.e. one not using the Venus OS Large image).

I have tried a complete new vanilla Venus OS Large installation on the Raspberry Pi. Same error. Removing the Victron VE.Direct USB made no difference. 

I have the latest version of the Venus software (as at 7 June 2022) which provides "Signal K 1.40.0-Venus". 

Any help fixing this issue would be most appreciated. 

Thanks for all your support for SignalK; it is a fantastic way of "mainstreaming" this software within the Victron community.

Thanks, 
Andrew

PS: Here is screen shot of the error:
[image]
PPS: It is amazing seeing Victron BMV-712 data on my Raspberry Pi screen; thanks for adding the Raspberry Pi support.

mpvader, 2022-06-08:
Hi amas008, we had the exact same question on Community yesterday, please add there / follow that. https://community.victronenergy.com/questions/138314/ngt-1-pi-and-venus.html

This issue tracker is for (our) development, not things line this.

thanks for the compliments, glad you like it!

amas008, 2022-06-08:
OK, thanks for the super quick reply; appreciated. I will follow the community link.