victronenergy/venus #961: Debian repositories in Wiki are incorrect and don't work

Drizzt321, 2022-06-19:
The documentation in the wiki at https://github.com/victronenergy/venus/wiki/install-venus-packages-on-Debian for trying to install to a barebones Debian install are very out of date. Also references needing Jessie version, which is very, very out of date, and doesn't appear to be in the packages repo/dir anyway. `dunfell` is, which isn't a Debian release at all. And all the files appear to be .ipk, not .deb as they should be. 

I'm guessing the Venus OS core, while still based on Debian Jessie, is effectively forked from that. If so, should probably just completely remove these instructions as being out of date and possibly even bad as they give someone an idea they can roll their own.

mpvader, 2022-06-19:
Hi @Drizzt321, that documentation is out of date indeed; I'll add a note.

We do still have Debian packages, .deb, but the last release version of Venus OS doesn't have them. So, latest available ones are here, see Debian-Buster and Raspbian-Buster folders:

https://updates.victronenergy.com/feeds/venus/candidate/packages/

closed 2022-06-19 by mpvader

Drizzt321, 2022-06-19:
@mpvader Ah! There it is, thanks. I'll see about giving that a try. Glad it's buster, not jessie too.

Dukat-Gul, 2022-07-19:
@Drizzt321 - this is how I have had it in the past.

```
#cat /etc/apt/sources.list.d/victronenergy.list
deb [trusted=yes] https://updates.victronenergy.com/feeds/venus/candidate/packages/raspbian-buster buster main
```

Head, 2023-04-01:
I'm trying to not have too many Raspberries running in my house. vrmlogger works fine in those packages, but I'm missing the while `gui` stuff. Seem to not be included in there? How can I install it?