victronenergy/venus #963: dbus_modbustcp: extra /System/ fields for ESS systems

dirkjanfaber, 2022-06-23:
In order to give the end user more information, I'd like to have the following fields added to dbus_modbustcp and node-red-contrib-victron.

- System/MaxCellTemperature
- System/MaxCellVoltage
- System/MaxTemperatureCellId
- System/MaxVoltageCellId
- System/MinCellTemperature
- System/MinCellVoltage
- System/MinTemperatureCellId
- System/MinVoltageCellId
- System/NrOfModulesBlockingCharge
- System/NrOfModulesBlockingDischarge
- System/NrOfModulesOffline
- System/NrOfModulesOnline

To be added to:
- [ ] dbus_modbustcp (ibu)
- [ ] node-red-contrib-victron (dfa)

jhofstee, 2022-06-24:
oh please don't do this.. This assumes there is a single battery.

izak, 2022-06-25:
You will have to explain it to me. If there are multiple CAN-bms batteries, they each get their own DeviceInstance and therefore their own Modbus slave address. And where there are multiple modules in a battery, the point is exactly that we want to know which module has the low/high cell, or how many modules are online/offline/blocking. I struggle to see how it would be affected by multiple batteries or modules?

Many of the mentioned items were already on modbus as well, and especially the online/offline/blocking items are extremely useful.

jhofstee, 2022-07-01:
> oh please don't do this.. This assumes there is a single battery.

Ah right, nevermind. I was in the impression you wanted systemcalc to handle this, not the BMS service itself.

dirkjanfaber, 2022-08-16:
Added to node-red-contrib-victron in https://github.com/victronenergy/node-red-contrib-victron/releases/tag/v1.4.21.

closed 2022-08-17 by mpvader