victronenergy/venus #965: python3: missing "dataclasses" module

smurfix, 2022-07-02:
```
# python3
Python 3.8.11 (default, Feb 15 2022, 23:17:33) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dataclasses
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'dataclasses'
>>> 
```
There is no `python3-dataclasses` OPKG module and the dataclasses backport from pypi only works with python3.6.

Please fix.

smurfix, 2022-07-02:
Same for "bdb" which you need e.g. for pytest

smurfix, 2022-07-02:
never mind, found the python3-modules package …

closed 2022-07-02 by smurfix

MaUwJo, 2024-07-02:
> never mind, found the python3-modules package …

what did you do in detail?
I wanted to run https://github.com/suaveolent/hoymiles-wifi but ended in the same error.

smurfix, 2024-07-02:
`opkg update && opkg install python3-modules`of course.