victronenergy/venus #968: USB RS485 adapter does not work properly in Node RED (v2.82-large-30)

Dermayer, 2022-07-13:
If you want to address a RS485 interface (/dev/ttyUSB1) in Node-RED this does not work reliably. 
Sending values seems to be no problem. When receiving values it seems that only every x. Message arrives. My guess is that Venus OS reads the interface empty. 
If you look at the LEDs on the USB adapter, you can also see that Venus OS sends something every x seconds, probably an autoscan for new devices.
I don't know a way to turn this off.

nmbath, 2022-07-15:
@Dermayer, 

Victron will close this issue and direct you to the community forums at community.victronenergy.com. This is for development tracking and not problems.

Search for Venus and serial-starter. You are correct that serial-starter will be trying to identify the device which is connected to the RS485. Do a search for serial-starter and Venus and you should find a load of material to help.

jhofstee, 2022-07-15:
Indeed, hint, have a look at /opt/victronenergy/serial-starter/stop-tty.sh. Closing indeed, since this works as intended.

closed 2022-07-15 by jhofstee

jhofstee, 2022-07-15:
for what it is worth, in udev rules you can add ENV{VE_SERVICE}="ignore" as well.