victronenergy/venus #971: unable to run conmanctl

georgelza, 2022-08-07:
Hi all
I recently purchased a Victron 500W 12V inverter/charge controller and a BMC from a friend (for my mom, I have a larger setup my side), he included a RPi3B with (as per him) this software. (when I asked him where he got it he indicated he deployed the image ....)

When I started everything I had the RPi hard wired into the router/switch and was able to access the information on VMR as he pre registered it for me. But as this entire setup is going to life on a trolly, I need Wifi working.

He pointed me to this repo, reading it seems I need to enable Wifi via conmanctl but executing that fails with command not found.
Izak mentioned that this is NOT build on Raspbian, so got me wondering as when I run uname -a I get Rasbian 4.19.81-u7 back.

please help

G

mpvader, 2022-08-07:
Hi, sorry, but this issue list is not for support. For support on Venus OS running on an rpi, please go to the Modifications section on Victron Community: https://community.victronenergy.com/spaces/31/mods.html.

Before you do, make sure that pi is running Venus OS. Sounds to me like its running Raspbian, which is not the same as Venus OS.

closed 2022-08-07 by mpvader

georgelza, 2022-08-07:
... the plan was more to get in touch with Mark as I referred to him, but could not find any other way to contact him.
thanks.
G