victronenergy/venus #977: Minor issue - Venus Large 2.90~14 will not allow update of Victron Nodes to 1.4.23

houser42, 2022-08-25:
Minor issue and I assume it will settle itself but for what it is worth:
Venus OS Large 2.90~14 will not allow update in "manage palette" to latest update of 
victronenergy/node-red-contrib-victron to version 1.4.23.
No idea if this are important fixes  in there worth the question. But there it is.

dirkjanfaber, 2022-08-25:
This is not a bug, it is a feature. We [made it impossible](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-extended/node-red-venus/files/venus-default-settings.js#L303) to update from the palette because the version of Venus and node-red-contrib-victron are tied together. 
Also see https://www.victronenergy.com/live/venus-os:large#qhow_can_i_update_the_victron_node_pallette

Usually I update the npm version of the package _after_ the release of a new Venus update, but in this case I was a bit too early with that.

closed 2022-08-25 by dirkjanfaber

houser42, 2022-08-25:
Understood! Thanks for taking the time to explain. Appreciated.