victronenergy/venus #980: cerbo gx stops to work

kvv515kvv, 2022-09-10:
Hi
cerbo gx stops to work. it is les then year old.
orange light is blinking no matter what i do
cerbo disapears from my phone 
touch screen is black

i checked all connections, turned off all inputs cables...
i tried several times power off power on... it still blinking orange
pressed push button short press, long press - no reaction ...it still blinking orange
powered with sdcard to make factory reset - same result - no reaction ...it still blinking orange
all other components are working fine (victron 3000, solar mppt, shunt)
any solution? or cerbo simply died and needs to be replaced with new one?

jhofstee, 2022-09-11:
This is not the right place to report such issues.

mpvader, 2022-09-11:
Hi @kvv515kvv sorry to hear you're having issues. Please either contact your dealer, whom you'll need anyway for repair/replacement. Or go to https://community.victronenergy.com.

Thank you, Matthijs.

closed 2022-09-11 by mpvader