victronenergy/venus #982: Dbus values not always displayed on GUI

AlbertronicDev, 2022-09-14:
After updating to v2.90, I (and several customer) notice that our BMS values are not always displayed anymore in the GUI. The values are on the dbus, but are not displayed. As soon as the value changes, the value is displayed in the GUI.

Dbus value example:
[image]

GUI result:
[image]

As soon as the value would change to a different value, for example 7, it is showing in the GUI.
Sometimes it is battery cell, sometimes SoC value, sometimes a different parameter. In any case when the value is changed, it starts displaying in the GUI.

Do you have any idea how this is possible? All BMS values are updated from running Python code every second:
 `self._dbusservice["/System/MinVoltageCellId"] = self.lowest_cell_voltage_num` but of course can contain the same value for a long time.

izak, 2022-09-15:
Hi @AlbertronicDev ,

It sounds as if `GetItems` is not implemented. The GUI will fetch all items from the service with one call. I see you use velib_python in your code. Therefore it should be sufficient to update velib_python.

GetItems was added to velib_python here: https://github.com/victronenergy/velib_python/commit/7062d839ce49007818b729bbc81cdb19a47faa55

AlbertronicDev, 2022-09-15:
Hi @izak ,
Thanks for the quick reply. I am not sure if I understand you correctly.
We indeed use velib_python in both python codes (smartbms.py and smartbms_manager.py), but just the latest version I suppose which is available on the GX. The code includes velib_python with the following line:
`sys.path.insert(1, os.path.join(os.path.dirname(__file__), '/opt/victronenergy/dbus-systemcalc-py/ext/velib_python'))`

Quick explanation: smartbms.py automatically is called by Victron when a FTDI USB is plugged in. For every USB instance, the smartbms.py is called with as argument, the TTY

smartbms_manager.py always runs, and combines all battery instances which are also SmartBMS, on the dbus. These combined values are then send back to the dbus as an extra battery. This one includes CVL, CCL and DCL.

Back to this issue:
We also noticed that this behaviour very specifically happens with VenusOS v2.90 and only because of the following code in each smartbms.py instance (which was working fine in previous VenusOS versions):
`def _is_com_available(self, port_match):
        available_ports = serial.tools.list_ports.comports()
        for port in available_ports:
            if port.device == port_match:
                return True
        return False`
        
Even more specific: as soon as the call to serial.tools.list_ports.comports() is done, the problems start occuring with values not being displayed. Even if the code calls comports() and always returns True, everything goes mad. As soon as I remove comports() and only return True, it works fine.

Not sure why this started happening.

jhofstee, 2022-09-15:
> Hi @AlbertronicDev ,
> 
> It sounds as if `GetItems` is not implemented. The GUI will fetch all items from the service with one call. I see you use velib_python in your code. Therefore it should be sufficient to update velib_python.
> 
> GetItems was added to velib_python here: [victronenergy/velib_python@7062d83](https://github.com/victronenergy/velib_python/commit/7062d839ce49007818b729bbc81cdb19a47faa55)

That should be irrelevant, the GUI will fall back to individual fetches.

AlbertronicDev, 2022-09-16:
Tried replicating it with a modified version of the code, which does not connect to a USB instance, but rather just sets static data into the battery parameters. Was not able to replicate it.
It only happens in a very specific situation in smartbms.py, where once per second, via Glib.timeout_add, the `serial.tools.list_ports.comports()` is called.
We fixed the situation by replacing the `serial.tools.list_ports.comports()` call with os.path.exists to check if /dev/ttyUSBx exists.

Still unknown how a call to `serial.tools.list_ports.comports()` let's the GUI not show values on the dbus unfortunately.

jhofstee, 2022-09-16:
I don't that either, but closing as resolved.

jhofstee, 2022-09-16:
And thanks btw, to actually fix issues here instead of raising new ones ;)

closed 2022-09-16 by jhofstee