victronenergy/venus #988: DVCC: Option to select which Multi(s) are controlled

izak, 2022-10-03:
This is for systems with more than one Multi/Quattro connected.

Presently the Multi with the lowest Device Instance that is switched on is controlled by DVCC. This is always the built-in VE.Bus port, which is a good thing.

This means that if you more than one Multi connected, and you turn off the "main" unit, DVCC starts controlling the second Multi. This is not really what you want.

Furthermore, if you later turn on the "main" Multi again, DVCC stops controlling the second multi and starts controlling the first Multi again. This will cause the second Multi to go into passthru, or switch off. Also not really what you want.

Finally, none of this is properly communicated to the user either.

Suggestion.
- [ ] Add configurable option to select which Multi is controlled by DVCC. Default behaviour is "Auto", the same as current behaviour.
- [ ] Show on the GUI which Multi/Quattro is used by DVCC
- [ ] Only selected Multi is published as `com.victronenergy.system/VebusService`, when not available, shows Invalid.

izak, 2022-12-15:
@warwickchapman , conversely we also have people with boats with multiple battery banks which may not even be of the same voltage. If I make you happy, I make them decidedly unhappy.

The point of this issue is to have a well defined state of operation. The present implementation is to control only one Multi, and changing that is a big departure. Flopping control back and forth between two Multis is however a problem that must be fixed, and the target of this issue.

Maybe in future we'll look into something like that again, but right now you will probably have to use Node-Red or similar to do it.

mpvader, 2023-05-05:
All done, adding the documentation part of this to the v3.00 documentation issue.

closed 2023-05-05 by mpvader