victronenergy/venus #989: VE.Bus BMS v2: Pre-alarm to VRM, GUI

izak, 2022-10-07:
The Pre-alarm for VE.Bus BMS v2 is available on dbus, but is not used anywhere nor shown to the user.

It should at least be logged to VRM, and probably shown on the "Alarm Status" page on the GUI.

TODO
- [ ] Log to VRM
- [ ] Add to modbus
- [ ] Show on GUI

mpvader, 2023-03-26:
@izak does this work in VRM, ie can you post a screenshot here of a pre alarm showing up as an entry in the VRM alarm log?

and can you make the required change in the gui code?

izak, 2023-03-27:
After a quick discussion with @vejpasop , we're going to make this an alarm so that it will show up in the Alarm log. At present it does not.

The dbus path changes tor consistency, so a vrmlogger update will also be required. Already requested a change in VRM as well.

vejpasop, 2023-03-28:
I added `Alarms/BmsPreAlarm ` the state of this can be 0 = OK, 1 = WARNING

Now when this pre alarm is active the Multi will also report al low battery warning. (`Alarms/LowBattery` = WARNING)

vejpasop, 2023-03-28:
Implemented in mk2-dbus v3.51

closed 2023-08-03 by izak