victronenergy/venus #994: When exporting to grid consumption is wrong

mr-manuel, 2022-11-06:
Hello,

I'm using Venus OS to integrate my existing PV (Enphase). To integrate it I'm using custom dbus drivers ( [venus-os_dbus-enphase-envoy](https://github.com/mr-manuel/venus-os_dbus-enphase-envoy) and [venus-os_dbus-mqtt-grid](https://github.com/mr-manuel/venus-os_dbus-mqtt-grid) ). All work fine, except when power is feeding into grid (exporting).

[image]

Debugging a bit I found that `_safemax` function in [dbus_systemcalc.py](https://github.com/victronenergy/dbus-systemcalc-py/blob/2f1609500c2aaa020c5d2bbcf811d57606eadccb/dbus_systemcalc.py) is preventing `consumption[phase]` to get negative. See this [lines](https://github.com/victronenergy/dbus-systemcalc-py/blob/2f1609500c2aaa020c5d2bbcf811d57606eadccb/dbus_systemcalc.py#L864-L865).

After removing `_safemax()` everything is like it should in my eyes.

[image]

So the question is, why this value is limited to be only positive?

Unfortunately I have not get my MultiPlus II-GX yet to test it with this change. If there is no reason I will open a pull request to remove `_safemax` from the lines mentioned above.

izak, 2022-11-07:
Hi @mr-manuel , We don't want a consumption number to be negative, because that would be nonsensical. That is why it is capped at zero. Consumption by definition has to be positive. Your suggested change is definitely not correct.

However, I think you simply have your PV-inverter incorrectly configured for your use case. Since you have no Multi in that system, some readings that the system needs to make an estimate is not there. Since your PV-inverter is configured to be on AC-out (I'm sure that is your mistake), the system makes the assumption that all PV is consumed by the loads directly.

It still sees the negative value reported by the grid meter, but has no way to know where that comes from so simply reports it as is.

If your PV-inverter shows itself as being on AC-In (`/Position` = 0), then you will see the numbers are correct.

I tested this before writing this. Here is a simulated system with a grid meter reporting 300W export, and a PV-inverter on AC-out making 1100W. Similar to your report, the entire 1100W is assumed to be consumption.

[image]

And here is the same setup where I changed `/Position` to 0 (AC-in).

[image]

I hope that helps!

closed 2022-11-07 by izak

mr-manuel, 2022-11-07:
Hi @izak thank you very much for explanation, this helped me a lot. Indeed now it's correct.

Is it normal, that the flow dots aren't displayed on the PV Inverter when the PV inverter is in position 0? On the grid they are displayed only if power is +/- 30W. Is there a higher thresholt for the PV Inverter and battery?

izak, 2022-11-07:
@mr-manuel the dots should always show when above 30W, and in testing it shows up at 31W.

[image]

[image]