victronenergy/venus #995: own driver installation: partly wrong documented?

schollex, 2022-11-14:
Hi,

I wrote my own serial-driver as documented in https://github.com/victronenergy/venus/wiki/howto-add-a-driver-to-Venus#3-installing-a-driver
The driver (= single python-script) works, but I have problems installing the driver, e.g.
there is no directory `/opt/victronenergy/your-program/service`,
instead it looks like there must be a directory `/opt/victronenergy/service/your-program`?

The documentation seems to be outdated - could someone check this.
I need information how to pimp up a single python-script to get a installed serial-driver.

I'm running VenusOS v2.92 (build 20221016213031) on RPi2.

mpvader, 2022-12-17:
Hi, were you able to get it working? I’ve for now added a comment to that documentation, saying that its out of date.

schollex, 2022-12-19:
> Hi, were you able to get it working? I’ve for now added a comment to that documentation, saying that its out of date.

I didn't give it a try.
But I would start transforming my tty-driver from manual tty-assignment to plug&play, if you could give me some hints.

schollex, 2023-02-20:
I noticed another related issue...
I updated Venus OS from v2.92 to v.2.93.
Now, root-mountpoint / is mounted as readonly-filesystem. Thus, modification on /etc/udev/rules.d/serial-starter.rules as described in https://github.com/victronenergy/venus/wiki/howto-add-a-driver-to-Venus is no longer possible.
My workaround: I execute /opt/victronenergy/serial-starter/stop-tty.sh with the proper TTY as parameter within the start-script of my own driver

mpvader, 2023-02-20:
Hi, the rootfs is read-only already per v2.70 or v2.80. See here for details on that: https://www.victronenergy.com/live/ccgx:root_access.

Looks like its chapter `Adding or modifying services` is the info you are looking for, and also exactly what we should update in the 'howto add a driver to Venus' document. We'll have a look at that, success and thanks!

dirkjanfaber, 2023-02-20:
Updated the documentation to match the current situation on Venus.

schollex, 2023-02-20:
> Hi, the rootfs is read-only already per v2.70 or v2.80. See here for details on that: https://www.victronenergy.com/live/ccgx:root_access.

Sorry for my wrong statement. During my previous firmware usage, I must have been executing `/opt/victronenergy/swupdate-scripts/remount-rw.sh`, but I can't remember :-(

closed 2023-02-21 by dirkjanfaber