victronenergy/venus #1000: vebus: Add Multi uptime counter

vejpasop, 2022-11-17:
The following dbus item will contain the Multi uptime counter `Devices/0/UpTime`.
This item will be updated every ~60 seconds.

Todo: 

- [ ] Add to mk2-dbus
- [ ] Add to GUI
- [ ] Add to VRM

vejpasop, 2022-11-18:
Included in mk2-dbus V3.43

mpvader, 2023-03-12:
@izak I see in vrmlogger that this logs the uptime for device 0 only; is that on purpose?

[image]

izak, 2023-03-13:
@mpvader That was a mistake. I incorrectly assumed this path counts the uptime for the whole "cluster", also because the dbus path is specifically mentioned above. I cleared it up with Jos, I'll have to add 11 more attributes!

izak, 2023-03-20:
After quick discussion with Jos, no need to add anything to the GUI. This is for debugging purposes, so we got everything we need already in v3.00~24.

Closing.

closed 2023-03-20 by izak