victronenergy/venus #999: vebus: Fix Multi subversion information and apply to FirmwareVersion

vejpasop, 2022-11-17:
Multi firmware versions can have additional sub version information to indicate special builds beta builds and unreleased X versions for testing,

For instance:

- Main release:  498
- Sub versions, derived form the main release  498.12 
- Special versions (designed for special application/customer) displayed as 498.s1
- Beta versions (pre-releases of upcoming version) displayed as 498(Beta 2)
- R&D version. (i.e. code is not archived) displayed as 498.X

Up till now this subversion information was only visible on the `FirmwareSubVersion` dbus path. (Due to a bug the text representation of the subversion was not always correct. Only the value was.)

The firmware subversion info will be added to the text representation of the `FirmwareVersion` item. This way the version will be automatically displayed correctly on VRM also.

[image]

Todo:

- [ ] Add to mk2-dbus

vejpasop, 2022-11-18:
Included in mk2-dbus V3.43

jhofstee, 2023-04-19:
@vejpasop can this be closed?

vejpasop, 2023-04-20:
@jhofstee Yes