victronenergy/venus #1001: Zigbee: Some DRF2658C modules not recognised

izak, 2022-11-22:
For some time now the newer CH340-based module is supported by this udev line:

```
# DRF2658C Zigbee converter with CH340 chip                                                           
ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_MODEL}=="USB2.0-Serial", ENV{VE_SERVICE}="cgwacs:default"   
```

At least one report from the field suggests that this may not work for some of them. Suggested to change the udev line to:

```
# DRF2658C Zigbee converter with CH340 chip                                                           
ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="1a86", ENV{ID_MODEL_ID}=="7523", ENV{VE_SERVICE}="cgwacs:default"
```

izak, 2022-12-20:
Included in 3.00~12. Tested on a Cerbo. Still detects the older DRF2658C correctly. No regressions in other words.

mpvader, 2022-12-20:
how urgent is it to get this out in an official release @izak ?

izak, 2022-12-21:
@mpvader It is not terribly urgent, but right now it is the case that Zigbees don't work out of the box, so probably not unimportant either. But I will mark it as a backport candidate so it can at least go out before Venus 3.00 if there is an opportunity.