victronenergy/venus #1002: WLAN password only accepts 35 character

jhaklotz, 2022-11-24:
When trying to enter my WLAN pw, I noticed that it is limited to 35 chars.
Unfortunately I require the full allowed 63 chars.

mpvader, 2022-12-09:
Hi @jhaklotz noted. We’ll look at this - someday. Its not a high priority for us

jhofstee, 2023-04-24:
Not a perfect fix for this, but at least it allows somewhat to enter such a long password:

```
change the maxlength for the wifi password to 63 characters

As reported here, https://github.com/victronenergy/venus/issues/1002, there are
users with really long passwords. This changes the gui to accept such a long
password. Entering such a long password can push the cursor of the screen, but
since having such long password is a rarity, no effort is spend to prevent that.
```

jhaklotz, 2023-04-25:
If cursor moves offscreen, that’s not nice but acceptable when such long password can be entered at least

closed 2023-04-25 by jhaklotz