victronenergy/venus #1006: Monitor separate hybrid inverter battery and support it in ESS

chaerle, 2022-12-09:
I have an existing hybrid high voltage solar inverter and want to to have to have a Multiplis II with his own battery to support the system in ESS mode. 

It should be possible the configure that, the this second battery is connected to the PV inverter instead of the Multiplus II and this should also be displayed on the main GX screen (see picture).

[image]

The ESS on the Multiplus II should monitor the charging power of the existing hybrid battery and take some configurable percentage to charge its own battery als long the inverter battery is not full. After that, it should take all surplus PV energy. It should also never charge its battery by discharging the inverter battery.

On feed in, it should ensure, that the hybrid inverter battery is not charged from the Multipus II battery, i,e, keeping not only the grid point but also the inverter battery charging power at zero. 

See also 
https://community.victronenergy.com/questions/171975/multiplus-ii-as-addition-to-existing-hybrid-invert.html

mpvader, 2022-12-09:
Hi @chaerle , that would all be very nice!

But I have to disappoint you, this is not something we’ll be working on anytime soon.

chaerle, 2022-12-09:
I step into this direction whould be just to support a second battery not connected to the Multiplus. Currently it seems to be that the Venus assumes that all batteries are conneted to the Multplus. As in the PV inverter options, where it is possible to select the connection point (AC-in or AC out), a battery should have the option to select, that is connected to the inverter, it least that is NOT connected to the multiplus. 

The extension to the ESS would be a separate thing it could also be done in Red-Node or with a python script.

There are lots of systems which have a separate battery system not unter control of the Multiplus, so the support for a second battery would open the usage of the Multiplus to plug in into these existing systems.