victronenergy/venus #1007: Add logic to enable restart after overload on mains failure for Quattro

KW4NP, 2022-12-15:
The Quattro advertises a 100 amp transfer switch. However, if using this transfer switch while the consumption is over the capacity of the Quattro (but still well under 100A) overload issues are very likely to occur because the inverter is used during the transfer. Using the inverter during the transfer is fine when the consumption is less than the Quattro can handle. Effectively the transfer switch is not functional in some conditions despite consumption being well under the rated and advertised 100A load.

Right now starting the generator on mains failure or overload because of mains failure is useless if the load was higher than the inverter overload limit because the inverters will have shut down due to overload by the time the generator has started **and do not automatically reset.** This leaves the generator running with the inverters in an error state and the load off, not good for unattended sites and generally not desirable behavior for the system. Many automatic transfer switches have a delay built in for the generator to start, so it seems reasonable to allow for a few minutes of outage and definitely preferable to a prolonged outage!

Is it possible to make a setting that will wait until the generator has started and then reset the system after an overload? Maybe as an option for the "Inverter Overload" generator start condition?

KW4NP, 2022-12-15:
From idea posted here[https://community.victronenergy.com/idea/175006/transfer-switch-mode-for-quattros-to-solve-overloa.html](url)

mpvader, 2023-03-25:
Hi @KW4NP the quattros will restart after 30 seconds. Unless the overload is of such extreme severity that the quattro thinks there is no other option that thats its broken.

so: are you sure they remain off? And in that case, can we take a look on the VRM system?

a workaround btw would be for you to use venus os large -> node-red and then the reset/restart system option we have

KW4NP, 2023-03-26:
Yes, I'm sure that they remain off. Have since installation, although
occasionally it will restart by itself. You're welcome to take a look at
VRM, just let me know how to give you access. I thought about using
nodered, but that's definitely a workaround and I don't want to add
complexity for the customer.

Thanks!

KW4NP, 2023-04-04:
@mpvader , how should I give you access to VRM to check into this issue?

KW4NP, 2023-05-09:
@mpvader it's been a while since I heard from you, are you still interested in looking at this issue?

mpvader, 2023-05-09:
Hey @KW4NP can you give me the name of such system on VRM? Or two names.

KW4NP, 2023-05-09:
LBIM MFLT CAL and CCDM are two that have had issues with not restarting under normal load. If you want the ID please send me a message.

KW4NP, 2023-05-25:
Have you seen the issues that are happening? In the last few days it looks
like it's overloaded several times.

mpvader, 2023-05-25:
Hi, no still not. I keeping wanting to look at this myself; but then it doesn't happen. And also now I have to go. I've asked someone else to look into this.

Sorry for the delay! Matthijs

KW4NP, 2023-05-25:
Where are you going?

mpvader, 2023-05-26:
away from the computer, into the plane. Its not relevant

izak, 2023-05-29:
@KW4NP I've been asked to look into this.

I tested this with a Multi (I can also do the test with a Quattro, but I can only do that tomorrow, and I doubt the result will be different). The Multi will restart after 30 seconds, three times in a row, and after that it will remain off. As long as your generator starts up in 90 seconds, this kind of lockout should not happen.

Even when the Multi has switched itself off: If AC power returns, the Multi will transfer the load to the AC-in within a few seconds. I tested this as well. My 3kVA Multi successfully transferred my 4kW dummy load to my grid connection.

So what I am saying is this should already work.

KW4NP, 2023-05-29:
Hello Izak,
I agree that this should work, and sometimes it does. But sometimes it
doesn't too, and there's another factor at play in this as well, because
I'm running a three phase system. Did you see the alarm log on VRM?

KW4NP, 2023-05-29:
Generator start/stop | Automatic monitoring | Generator not detected at AC-input: Alarm | 2023-05-28 12:38:59 | 9m, 33s
-- | -- | -- | -- | --
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-28 12:35:54 | 19s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-28 12:35:47 | 7s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-28 12:34:23 | -1m, 21s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-28 12:33:23 | -22s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-25 18:52:58 | 8s
  | VE.Bus System [276] | Automatic monitoring | Overload L2: Alarm | 2023-05-25 18:51:38 | 37s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-25 18:51:38 | 1m, 20s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-25 12:18:40 | 13s
  | VE.Bus System [276] | Automatic monitoring | Overload L3: Alarm | 2023-05-25 12:16:10 | 2m, 30s
  | Generator start/stop | Automatic monitoring | Generator not detected at AC-input: Alarm | 2023-05-24 19:05:30 | 9m, 51s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-23 18:26:26 | 8s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-23 18:25:16 | 1m, 10s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-22 20:53:12 | 6s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-22 20:50:49 | 2m, 23s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-22 15:10:44 | 11s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-22 15:06:46 | 3m, 58s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-20 16:53:08 | 8s
  | Generator start/stop | Automatic monitoring | Generator not detected at AC-input: Alarm | 2023-05-20 16:52:13 | 1m, 20s
  | VE.Bus System [276] | Automatic monitoring | Overload L2: Alarm | 2023-05-20 16:47:20 | 22s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-20 16:47:20 | 5m, 48s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 10: System time synchronisation problem occurred | 2023-05-16 17:33:32 | 7s
  | VE.Bus System [276] | Automatic monitoring | Overload L3: Alarm | 2023-05-16 17:32:53 | 22s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-16 17:32:53 | 39s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-16 17:29:21 | 9s
  | VE.Bus System [276] | Automatic monitoring | Overload L3: Alarm | 2023-05-16 17:29:00 | 21s
  | VE.Bus System [276] | Automatic monitoring | Overload L1: Alarm | 2023-05-16 17:29:00 | 21s
  | Generator start/stop | Automatic monitoring | Generator not detected at AC-input: Alarm | 2023-05-16 17:26:46 | 3m, 1s
  | VE.Bus System [276] | Automatic monitoring | VE.Bus Error: VE.Bus Error 3: Not all, or more than, the expected devices were found in the system | 2023-05-16 17:23:09 | 7s

KW4NP, 2023-05-29:
As you can see, the issue seems to be that one phase goes down, which then brings a different error, which might be what locks out the generator. Sometimes I have to manually reset, and other times it comes back by itself. I'm guessing that the three phase system is the difference between our systems.

izak, 2023-05-29:
@KW4NP I agree, I think something else is wrong there. I'll get someone from support to contact you. I'm closing this issue as this is not a Venus-related issue.

closed 2023-05-29 by izak