victronenergy/venus #1009: Beaglebone Black Cape Firmware MK3

Tomi87, 2022-12-16:
Hello, I would like to ask if anyone has the firmware for the PIC18F24K22 Beaglebone Black cape MK3. I would like to make a Venus GX Beaglebone Black. I would happily buy the firmware if someone had it.

mkrenzke, 2022-12-16:
Hello, the firmware is not available to the public. Anyway there is a way to get around.

Since this is not a development-related issue, please ask your question in the [Community](https://community.victronenergy.com) again. Thanks.

mpvader, 2022-12-17:
I recommend to use a MK3-USB. And use that as is. All other options, like taking an MK3-USB and soldering the MK3-USB off of it to use that is prone to issues since there are small nuance differences that nearly nobody knows about; and the people that do need to develop new product rather than help DIY builds of a GX Device.

Sorry to disappoint but I hope you understand, all the best!

For any further questions about this, please use the Modifications section in Community: https://community.victronenergy.com/spaces/31/mods.html

closed 2022-12-17 by mpvader

Tomi87, 2022-12-17:
could you at least help me configure the 7 inch HDMI touch monitor for venus os beaglebone black? Venus os is already installed, but the LCD display is not working for me.

mpvader, 2022-12-17:
I don’t know how thats done; sorry.