victronenergy/venus #1011: Devices always get disconnected

jjtoard, 2022-12-16:
Since I added a second bms and second smartsolar controller my rapsberry pi3 started crashing (at least no network connection at all)
I bought a new raspberry pi4 4gb and now my devices get disconnected all the time.

Anyone having the same issues ?
How to solve it ?

My devices are connected via the DIY rs485 to usb connector cables but it is strange that they work when I reboot but get disconnected sometimes after a few minutes sometimes after some hours.

thepaulcooper, 2022-12-16:
I have a similar issue that I put up on the community site a few hours ago.  Please take a look at: https://community.victronenergy.com/questions/175242/multiplus-ii-gx-regularly-rebooting-and-causing-mp.html

My MPPT disappears periodically until I reboot.  The disappearance coincides with the Multiplus II GX rebooting itself for no apparent reason, but not every time.

jjtoard, 2022-12-16:
I don't think I have the same issue.  I don't have the feeling my multiplus is rebooting but not 100% sure.
The setting "reboot device when no contact" is disabled at my side.

thepaulcooper, 2022-12-17:
That was my belief at first too.  My MPPT just kept disappearing until a manual reboot of the Multi brought it back online.  But on further investigation, I found that it was regularly rebooting.  I first noticed this on the Multiplus II GX's LCD screen.  I verified this by pinging it every 2 seconds  - on checking the output there were periods of 'timeout'.  I then checked the Multi's messages log and found frequent reboots (but only during daylight hours!).  It's a 'soft' reboot as the device appears to continue to function normally and there is no rattling of the relays that you get when you power cycle it.

mpvader, 2022-12-17:
Hi @jjtoard , for troubleshooting like that, please use the Modifications section on Community: https://community.victronenergy.com/spaces/31/mods.html. This issue tracker is for development, not troubleshooting. Thanks!

Hi @thepaulcooper, that (partially) sounds like a known issue related to the hardware, I'll answer you on Community.

I'll close this issue, have a good weekend.

closed 2022-12-17 by mpvader