victronenergy/venus #1012: PV Limit set to Max when ESS is Mode3.

shaneyake, 2022-12-19:
The PV power limit is always getting set to Max and over writing my control, even tho ESS is set to Mode3.

https://community.victronenergy.com/questions/171543/fronius-pv-limit-always-going-to-max.html

izak, 2022-12-20:
@shaneyake How often do you write the limit?

The limit resets to default every 60 seconds. This is built not only into our software, but also into the sunspec protocol. When you send a limit to the PV-inverter, you also tell it how long that limit is valid, and in our implementation that is 120 seconds.

Therefore the PV-inverter will always reset after 60 to 120 seconds, and ideally you should send the limit every 30 seconds.

Can you confirm that you send the limit often enough?

shaneyake, 2022-12-20:
I am using Nodered to write to the Dbus every 2 seconds via Victron PV Inverter Node. I have tried 10, 5 and now 2 seconds with no change.

Since I am writing to Dbus I am assuming that your implementation takes care of the rest and I just need to write to "Ac/PowerLimit"

Can you confirm that when ESS is in Mode3 that nothing else is writing to "Ac/PowerLimit" ?

Thanks

shaneyake, 2023-01-14:
I was running 2.92 as it was listed as the latest stable release. Switched to V3.00 ~ 14 and no longer seeing my control being overwritten. Everything works now, this was information I was looking for.

Could not get the dbus-monitor to output a sender id. This is what I got.
```
# dbus-monitor --system path=/Ac/PowerLimit,member=SetValue
signal time=1673636111.575667 sender=org.freedesktop.DBus -> destination=:1.71623 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.71623"
```

izak, 2023-01-16:
Thanks. So maybe you ran into the earlier bug that is already fixed. That explains why dbus-monitor also shows nothing. I will close this issue then.

closed 2023-01-16 by izak