victronenergy/venus #1015: v3.00 bug: PV Inverter on AC in disappears after update

mpvader, 2022-12-26:
Reported on Community, in tilde-14 as well as previous thread:

The Ac_In Fronius dissapears from both GX page and from VRM side. Rolling back to 2.92

https://community.victronenergy.com/questions/164781/venus-os-v3004-available-for-testing.html?childToView=176324#comment-176324

jhofstee, 2023-01-12:
thanks, that removes one of my todo things (to tell you this), this is indeed weird since indeed nothing has changed Fronius wise between mentioned version and both VRM and the GUI are effected, so it likely not a GUI issue either....

izak, 2023-01-13:
Heisenbug. Adding an observer (me) made the bug go away. Upon rebooting it this morning, it works perfectly. Even rebooting it to make sure yields the same result.

For reference, the original event happened at `2022-11-08 07:16 UTC`, when he upgraded to v3.00~4. It doesn't look as if there was an attempt at rebooting to "fix" it. By now many of the log files have long moved on, but the systemcalc log goes back far enough that I can see the `pvinverter` services DID come up.

```
2022-11-08 07:17:19.164410500 INFO:dbusmonitor:Found: com.victronenergy.pvinverter.pv_232_1939230, scanning and storing items
2022-11-08 07:17:19.170436500 INFO:dbusmonitor:       com.victronenergy.pvinverter.pv_232_1939230 has device instance 21
2022-11-08 07:17:19.192317500 INFO:dbusmonitor:Found: com.victronenergy.pvinverter.pv_102_285801, scanning and storing items
2022-11-08 07:17:19.198082500 INFO:dbusmonitor:       com.victronenergy.pvinverter.pv_102_285801 has device instance 20
```

Logs for the GUI, and vrmlogger doesn't go back far enough to make a call as to why they would not have seen the new service.

On 2022-12-22 he upgrades to ~14, around 23:24 UTC. Again, systemcalc shows the pvinverters coming up:

```
2022-12-22 23:24:58.408264500 INFO:dbusmonitor:Found: com.victronenergy.pvinverter.pv_232_1939230, scanning and storing items
2022-12-22 23:24:58.420850500 INFO:dbusmonitor:       com.victronenergy.pvinverter.pv_232_1939230 has device instance 21
2022-12-22 23:24:58.436622500 INFO:dbusmonitor:Found: com.victronenergy.pvinverter.pv_102_285801, scanning and storing items
2022-12-22 23:24:58.450794500 INFO:dbusmonitor:       com.victronenergy.pvinverter.pv_102_285801 has device instance 20
```

Again, he spent only a few minutes there before booting back to 2.92.

Unfortunately, due to the EV-charger constantly coming and going, the vrmlogger log file is spammed out all the way to 11 January.

VRM shows data for the PV-inverter(s) during both events, so it also doesn't look like vrmlogger missed the presence of the PV-inverters.

It seems to come down solely to the GUI now, and we only have a report for 3.00~4.

izak, 2023-01-13:
Bingo, the customer brought up some detail and this one is solved. It is because of [this bug][1] that was fixed.

His PV-inverter was on AC-out-2, but configured as being on AC-out-1 (genset). After the mentioned fix kicked in, his PV-inverter moved to the genset side, and disappeared off the overview. Correcting the configuration made it show up again.

[1]: https://github.com/victronenergy/venus/issues/906

izak, 2023-02-20:
Need to look into a migration for this, ideally.

Step 1: Find out how many people actually have a PV-inverter configured for AC-out-2 (previously assumed to be a genset). We won't be migrating these.

Step 2, Migration:
* If PV-inverter is on AC-in-1 (formerly grid), and
* AC-in-1 is configured as generator, and
* AC-in-2 is grid, then
* Move PV-inverter from AC-in1 to AC-in-2.

mpvader, 2023-02-20:
I can also just mention this in the release notes.

mpvader, 2023-03-25:
@izak pls make a proposal of few lines of text to go into the release notes about this. I like that much better than making some conditional migration script.

izak, 2023-04-17:
Added to the main TODO, under the vrmlogger v2.259 & v2.260 heading when it was changed.

mpvader, 2023-05-05:
ok I've seen it. good.

closed 2023-05-05 by mpvader