victronenergy/venus #1016: Generator start/stop: generate alarm when generator is not in autostart mode

jepefe, 2022-12-27:
- [ ] Add alarm to generator start/stop (jpe)
- [ ] Add alarm to Venus gui (jpe)

To prevent people from disabling autostart and then forgetting to enable it. 

Alarm is optional and can be enabled from generator start/stop settings.
Alarm should wait 5/10 minutes before triggering, this lets the operator some time to do whatever he has to do.

mpvader, 2023-03-25:
@jepefe & @izak how about making this part of the generator start stop changes now being worked on?

and then also include a resettable downwards counting timer?

i’ve been thinking about that, and only sensible solution is downward counting service timer.

not upward counting.

kwindrem, 2023-03-26:
Interestingly, I added a service timer as part of GuiMods. Might check that out.

My counter actually counts up but it's displayed as time remaining until next service. It was easier to handle it as an up count especially when changing the service interval.

mpvader, 2023-05-09:
Moved; too late for v3.00

jepefe, 2023-05-26:
Finished.
gui: [feat_service_interval](https://github.com/victronenergy/gui/tree/feat_service_interval/qml) 
dbus_generator: [v1.6.2](https://github.com/victronenergy/dbus_generator/releases/tag/v1.6.2)
Ready to be added to the 3.10 ToDo list once it is open.
[image]
[image]
[image]

mpvader, 2023-08-22:
can be closed @jepefe ? It works well, I used it :)

jepefe, 2023-08-22:
Yes, can be closed. Thanks for testing it :)

closed 2023-08-22 by jepefe