victronenergy/venus #1024: ESS: Improve options for peak shaving in self consumption modes.

izak, 2023-01-17:
## Description

This does not affect systems where the ESS mode is set to "Keep batteries charged".

In systems running in Optimised mode, the inverter component/bridge is disabled when we reach the configured SOC limit. Another way of saying this: Feed-in of energy is entirely disabled when MinSoc is reached.

We do this, because in this configuration the Multi uses the least energy, which is what you want when the battery is at the minimum SOC.

While feed-in is disabled, the Multi cannot observe the AC input current limit. If you exceed the AC input current limit, the Multi will go into passthru and simply take the power from the grid.

This has two bad side effects.

1. Systems where peak shaving is desired stops peak shaving at MinSoc.
2. It could trip a breaker.

## Suggestion
Instead of having a hard cut-off at MinSoc, allow a configurable window within which self-consumption is disabled, but where energy feed-in remains enabled so that the AC input current limit can be observed and peak shaving will work.

Something like this: 
```
100%
   |     <---  Capacity reserved for self consumption / ESS
60%
   |     <---  Capacity reserved for peak shaving
10%
   |     <---  Reserve capacity, only used during grid outage and used to prevent battery from switching off.
0%
```

## Caveats
* BatteryLife: This increases the MinSoc by 5% each day that a full charge (85% or more) is not achieved. The peak shaving limit will have to move with the "active soc limit" in some manner.
* Backwards compatibility. The feature should be disabled by default, so existing systems don't suddenly behave differently.

ajaxharg, 2023-01-29:
Another thing that would be useful is if the system could dynamically adjust the charge rate from the grid during a scheduled charge.  ie rather than charge at the maximum allowed limit instead charge at a rate which would hit the SOC at the end of the charge period.  This would allow the charger to potentially operate in a more efficient power band reducing wastage of energy from the grid.

RandMental, 2023-01-29:
Hi Andrew 

I would be interested in getting the efficiency curves for the charger in the Multiplus 48/5000  unit.

Can you help?

 

Regards

Rand

izak, 2023-01-30:
As always, the graphical part is the hardest. Suggestion below.

As always, we use a selection rather than a switch, because that allows descriptive text that explains what is happening. So we introduce a setting that regulates how the AC input limit is handled. It has two options: Above minimum SOC, or Always.

This setting is available (and used) only when in Optimised mode. In "Keep batteries charged" the AC input limit is already always observed.

[image]

[image]

An informational message explains why you'd want to use such a setting. If you set it to "Above minmum SOC", which is the current behaviour, you get:

[image]

And if you select Always, you get this warning:

[image]

mpvader, 2023-02-15:
Hey Izak, here are new texts for those two toasts:

First message:

> Use this option in systems that do not have a peak shaving function.

Second message:

> Use this option in systems are intended for peak shaving.
>
> Set the AC input current limit to configure the peak shaving threshold. Note that an energy meter - if installed - is not used for peak shaving. Only for ESS. See documentation for further information.

Its a difficult pair of sentences, welcome to improve further.

Especially the "Note that an energy meter" ... sentence. What I want to do their is make clear that peak shaving doesn't .

Since thats what most people will want: peakshaving at the utility connection. Not only be peakshaving the critical loads.

Still this change is a nice step in the right direction; at least for systems without energy meter we can now do proper peakshaving!

izak, 2023-02-16:
[cid:c919838d-aab2-4481-a361-14ba3e4cf673][cid:c9834349-f120-4d75-9919-750ff2572157]

closed 2023-02-16 by mpvader