victronenergy/venus #1026: Bluetooth-Stack crashes in client mode after reproducable amount of time/data

baranator, 2023-01-19:
Hi!

I am using a JK BMS in combination with the dbus-serialbattery service on my raspberry pi 3 with Venus OS 2.92. As my units serialport seems to be broken i started an enhancement to make dbus-serialbattery support its bluetooth connection which already works quite well (see https://github.com/Louisvdw/dbus-serialbattery/pull/372 )

Unfortunately during testing me and others noticed reproducable crashes of the whole bluetooth-stack. After the crash the bluetooth-adapter does not appear in `bluetoothctl` anymore and the only way to make it usable again is a full reboot or the following commands:

```
kill -9 $(pidof bluetoothd)
rfkill block bluetooth
rfkill unblock bluetooth
/etc/init.d/bluetooth start
```

we tried to track this down and noticed that we get this error almost always after 15h22m of receiving data from the bms. we noticed:

>interesting. Most reassembled Frames that contain the cell-info are 300 Bytes, some other are up to 320 Bytes long. Thosse are sent every second, giving 55340*300=16.602.000 Bytes. That's suspiciously close to 16.777.216 (16MB), which is 2^24, which is the highest number that can be represented as 3 Bytes unsigned. Maybe there is a counter/buffer overflowing somewhere in the BT-driver? 

this might be random but looks suspicious. To reduce the possible causes i installed a minimal archlinuxarm on the same raspberry pi 3 and ran the scraping of the jkbms-data via bluetooth manually without any venus-os-stuff and it now runs for 4 days and hasnt had a crash yet.

So at the moment we have the suspicion that the cause of the problem lies somewhere in the Venus OS code, maybe in the old bluez-firmware that is being used? 

Any ideas on this?

mpvader, 2023-01-19:
Hi!

Yes, my suspicion would be the same: old bluez. Or perhaps older kernel or older driver somewhere. What you are doing has no specific interaction with any Victron-written code. At least I cannot think of any.

But: updating bluez or otherwise trouble shooting this issue is not something I expect to ask anyone here anytime soon; other priorities I’m afraid.

Perhaps its possible to replace bluez without doing a full rebuild. See our wiki pages on SDK. And otherwise if you really want to go for this (consider buying a new rpi3) you could try get into building Venus OS. A while ago it was possible to build most of it even without having access to the private part of the sources.

but we never try that ourselves, hence might no longer work. Ans also doing anything with OE has proven to be a really steep learning curve, and lastly we will be very limited in supporting anything. Have to keep focus.

hope that helps anyway!

lastly: is the comms with the jkbms via bluetooth done using a real bluetooth connection? Or is it done by picking up broadcasts/advertisements?

mr-manuel, 2023-01-20:
May you could provide us more informations with which options the BlueZ driver was compiled for Venus OS?
I saw that the driver in Venus OS 3.00-14 is still 5.55. So it's not planned to update it to the latest version with the latest version of Venus OS?

mpvader, 2023-01-23:
Hi, indeed we don't have any update planned at the moment.

On Bluez we apply at least one patch; and how exactly its compiled requires a bit of a search through code.

Here is a bit of it:
https://github.com/victronenergy/meta-victronenergy/tree/master/meta-venus/recipes-connectivity/bluez5

and here is more:
https://github.com/victronenergy/openembedded-core/tree/dunfell_venus/meta/recipes-connectivity/bluez5

and upstream is here:
https://github.com/openembedded/openembedded-core/tree/dunfell/meta/recipes-connectivity/bluez5

or here in a newer branch:
https://github.com/openembedded/openembedded-core/tree/master/meta/recipes-connectivity/bluez5

BUT: here the responses really stop I'm afraid. As said, OE (Open Embedded) has a steep learning curve, and above starts to show why. To build and test a newer BueZ takes hours; and with a bit of bad luck you suddenly find yourself on a path that takes days to complete.

So, while I'd love to help and explain more; we cannot/should not.

closed 2023-01-23 by mpvader