victronenergy/venus #1031: Vebus: Improve generator on AC in 1 start/stop handling

vejpasop, 2023-02-06:
The general recommendation is to wire the generator to AC input 1. The main reason is that when the battery is completely empty or disconnected, a Quattro can only start from AC input 1 and not AC input 2.

Currently for the generator start/stop option to automatically stop a backup-generator once the grid has returned after an outage the generators is has to be connected to AC in 2

This can be fixed by adding an option to automatically stop the generator when connected to AC input 1. This change will include a feature to first disconnect from the generator, and only thereafter stop it. Making for a much smoother hand-over from generator to inverter mode. 
Additionally a (configurable) cool-down and warm up period should be implemented.

### todo 

- [ ] vebus frimware: Add ability to separately control ignore ac inputs and read back Ac in availability
- [ ] mk2-dbus: Add new dbus items and remove default "generator connected" setting from AC input 2
- [ ] generator-start stop: Implement required functionality
- [ ] VRM: Add new dbus items to VRM 

### Mk2-dbus

The following dbus items will be added

- `/Ac/Control/IgnoreAcIn1`
- `/Ac/Control/IgnoreAcIn2`
- `/Ac/State/AcIn1Available`
- `/Ac/State/AcIn2Available`

### Related issue
 https://github.com/victronenergy/venus/issues/303

vejpasop, 2023-02-07:
Added to mk2-dbus v3.48

vejpasop, 2023-02-10:
Added to Multi firmware version 205.
mk2-dbus v3.49 is on the Venus todo.

sdikiy, 2023-03-07:
It may be worth adding a delay (which can be changed) between when ACin1 or ACin2 is available and when the generator stops.

KW4NP, 2023-03-24:
I agree that a delay for changeover might be useful, especially for some of the more unstable grids it's really annoying to have the generator kick on repeatedly. But we're really looking forwards to this feature, so don't delay it to add a delay!

jepefe, 2023-05-08:
I reviewed the code and also ran some tests and all that was working before remains working. Also tested warm-up and cool-down feature, works good too.

mpvader, 2023-05-09:
@izak pls initiate the vrm work; once done you can close this issue

mpvader, 2023-05-09:
Also; does it make sense to have these new mk2dbus in nodered? 

And are they documented on the public dbus wiki page? Pls make sure they are; incl that they are used by gen startstop since v3.00; thanks!

mpvader, 2023-05-12:
Any remaining things here are for v3.10

Wholistic, 2023-07-03:
A bug report from a large user in the field, 

Site has a large 110 kVA generator with its own shutdown circuit which is not working as expected with the Venus generator cooldown timer and 9 x 15 kVA quattros.

Even if they set the controller shutdown to 0 seconds, it still takes 5 seconds or so for the generator to shut itself down after receiving the shutdown signal from the GX relay.

If they set the GX cooldown to 60 seconds, when the Victron cooldown engages it disconnects the AC-input from the generator, then after the 60s time period opens the relay to send the signal to stop the generator.

This sends the generator into its own cooldown mode, which is not immediate, taking 5+ more seconds. 

At this point the Quattro still detects AC available on the AC input and then Quattro reconnects, drawing full load 110 kVA for a few seconds.
Then the generator self-shutdown timer finishes and turns itself off.

The opposite of the intended smooth cooldown. 

The customer requests that there is an AC-input relay time limit after the cooldown timer before the generator would be allowed to reconnect (prevent immediately re-closing the AC-input relay again).
 
They believe this current behaviour may have just burned out a feedback relay on their Quattros setup. 

Customer can be contacted at support@myenergysolar.com.au

kwindrem, 2023-07-04:
I had a user of GuiMods and ExtTransferSwitch report the same problem today. In his case the generator produces voltage for about 2 seconds after the GX relay opens and the AC detection logic in the Multi/Quattro is faster than that!

I think the correct solution is to add a user configurable "post cool-down delay" that extends the ignore AC input for that period after cool-down ends and the generator is stopped.

Is this an issue which Victron intends to address rapidly? If not, I'll make an addition to my code until it does get implemented.

mpvader, 2023-07-04:
Hi both, thank you for reporting.

> Is this an issue which Victron intends to address rapidly? If not, I'll make an addition to my code until it does get implemented.

yes.

does it need to be configurable? I’d prefer adding a hardcoded number; like 10 seconds or something.

kwindrem, 2023-07-04:
My only thought was if a generator had it's own cool-down period that was not adjustable, any hard coded delay used in the GX device might not be enough.

If any generator cool-down period can be set to zero or a few seconds, then a hard coded value should be sufficient. I'm doing exactly that right now for GuiMods/ExtTransferSwitch. I picked a delay of 15 seconds. I'll change that to whatever fixed delay Victron decides on or pick up a user-configurable value if that happens.

Wholistic, 2023-07-04:
In this case a 30 second hard wired delay before allowing ac input again would resolve it, and be the fastest remedy as it wouldn't require any extra UI work. 

I think that's the best fix, and then see if any additional issues emerge before putting any more time into it.

mpvader, 2023-07-17:
Per v3.10~17 we added a 15 second hard wired delay.

mpvader, 2023-08-30:
Closing.

Most of this issue was implemented in v3.00 by the way.

and some bug fixing in v3.10

Wholistic, 2023-11-06:
A new bug report for a customer where the hard coded 15 seconds was not long enough. 
https://community.victronenergy.com/questions/240918/need-multiplus-to-ignore-generator-ac-during-cool.html

mpvader, 2023-11-06:
Continues in a new issue: https://github.com/victronenergy/venus/issues/1171