victronenergy/venus #1032: Vebus: Add BMS Connection lost alarm

vejpasop, 2023-02-13:
Add an on alarm on BOL frame timeouts. 
See https://www.victronenergy.com/live/mppt-error-codes#err_67_-_bms_connection_lost

Multi firmware >= 502 supports a BOL frames timed out status flag. 

Todo:

- [ ] Add to mk2-dbus
- [ ] Add to GUI
- [ ] Add to VRM

vejpasop, 2023-03-28:
Indeed the the low voltage battery will be shown along with the BMS comms lost.
Suppressing this  alarm in vebus-interface is not really favorable (in my opinion) . 
The low battery alarm will usually come up before the comms lost. And after the comms lost it takes a while (>20 seconds) to clear the low voltage alarm, afterwards it will go through a low battery warning. This has to do with the charge engine state in the Multi.

closed 2023-04-20 by mpvader

vejpasop, 2023-06-01:
Correct, 

Only it is note error 67 but a /Alarms/BmsConnectionLost

mpvader, 2023-06-01:
Can you share me screenshot of how that looks in the gui?

vejpasop, 2023-06-02:
[image]

warwickchapman, 2023-06-02:
Have tested this with site 308284.

### GX
Product | C014
Firmware version | v3.00~42

### VE.Bus
Product | MultiPlus-II 48/5000/70-48
Firmware version | 502

I unplugged the the VE.Bus cable from the GX device to simulate GX failure and after 30 minutes, nothing happened in 30 minutes.

I unplugged the BMS CAN cable from the Cerbo for the BYD BMU and after about 4 mins 30 seconds the BYD BMU shut the batteries down.  The Cerbo reported a low battery and then continued running for an hour with the DC bus energised by the AC Input.  I reconnected the BMS CAN cable and fired the batteries up again:

[image]
[image]

I will try this on another system with Freedom Won and report.

vejpasop, 2023-06-02:
Number 1 can be explained:
The BMS connection lost alarm in case of a managed battery is triggered by the Multi. So the Multi tells the GX device that it did not receive battery management messages for at least 5 minutes,   If you unplug the VE.Bus cable from the Multi the GX device can not read the status and will therefore not show the BMS connection lost.

Number 2 where you disconnect the CAN cable should have triggered a BMS connection lost. 
I will look into that.

vejpasop, 2023-06-02:
@warwickchapman Found it. I looked on your site. You need to update the MK3 in order for this to work.