victronenergy/venus #1039: systemcalc: Add summary path for inverter/vebus DC current

izak, 2023-03-09:
At the moment the GUI (and VRM?) gets the VebusService from systemcalc, and then queries it directly to find /Dc/0/Power, to animate that line on the Overview.

This does not work for Multi-RS. What we need is a summary path showing the inverter/inverter-charger DC current and power.

Probably something like `/Dc/Inverter/{Power,Current}`

mpvader, 2023-03-09:
why doesn't that work for the RS?

izak, 2024-04-05:
The balls-view now animates the DC to Multi-RS (and also other inverters including Inverter RS).

[image]

izak, 2024-07-29:
After a short discussion with Alex, no need to log this to VRM.

closed 2024-07-29 by izak