victronenergy/venus #1041: Building VenusOS for Libre Computers le Potato

thucar, 2023-03-13:
Is there a guide on adding a new machine config to build VenusOS for?

I would love to be able to run VenusOS on my lePotato boards. At first I had hoped that VenusOS was a version of Raspbian so I could use this script to achieve the desired result: https://github.com/libre-computer-project/libretech-raspbian-portability

Unfortunately as this is not the case,  it seems I would need to build VenusOS specifically for lePotato.

mpvader, 2023-03-13:
There is a simple guide here:

https://github.com/victronenergy/venus/wiki

but beware, what you are going to do is in most cases weeks of full time work when experienced. And without experience far far more.

and from our side there is no to very little time to help.

mpvader, 2023-03-13:
Hi, I don't sorry. And if I knew someone who could do this efficiently, most likely I'd have some priorities of my own for him or her :)

closed 2023-03-13 by mpvader