victronenergy/venus #1043: How to increase limit for open files?

tmuellerchen, 2023-03-18:
When I use Node-Red Dashboard I often got this message: EMFILE, too many open files.
I check with ulimit -n and it shows "1024"

How to increase this?

I work with 

raspberrypi2 5.10.110-rpi-venus-3

On a Raspi 3

Thanks in advance

mpvader, 2023-03-25:
Increasing the open files limit would need to be done in the daemontools
run script for the node-red service before switching to the nodered user.

Which in our source code is [here](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-extended/node-red-venus/node-red-venus.bb#L20) , but on your device its most likely a file in `/services/node-red` or some similar name.

mpvader, 2023-03-25:
@dirkjanfaber I think its a good idea to change this in general; agree?

Then can you make a change, test it, and then add it on the todo?

dirkjanfaber, 2023-03-26:
With so many files open, it might be wise to take a closer look at the efficiency of the used flow and why so many files are open first. As far as I know, this is the first user complaint where the limit gets hit. I checked at several systems running Node-RED (with dashboard) and, run `ls /proc/$(pgrep node)/fd/  | wc` to see the number of open files.  It usually is between 23 and 40 open files, so hitting the 1024 limit could be caused by a misbehaving node. 

@tmuellerchen: Can you report which packages you have installed via the package manager and possibly share the flows of the install that causes this issue?

dirkjanfaber, 2023-12-20:
Closing this issue due to inactivity.

closed 2023-12-20 by dirkjanfaber