victronenergy/venus #1049: mDNS name does not correspond to system name and is unpredictable if there is more than Cerbo in the network

hsteinhaus, 2023-03-23:
The Cerbo currently announces itself as "venus.local", regardless which name is set [Settings -> System Setup -> System Name]:
```
=  wifi0 IPv4 Victron                                       _http._tcp           local
   hostname = [venus.local]
   address = [10.22.20.56]
   port = [0]
   txt = ["Victron=true" "generic=true" "UrlPath=app"]
```

As far as there is only one Cerbo present in the network, this is only a convenience issue. When a second system is present in the same subnet, the second Cerbo claims the mDNS name `venus-2.local`. Unfortunately, the order is completely random, based on boot order:
```
=  wifi0 IPv4 Victron #2                                    _http._tcp           local
   hostname = [venus-2.local]
   address = [10.22.20.31]
   port = [0]
   txt = ["Victron=true" "generic=true" "UrlPath=app"]
```
This is very nasty problem e.g. for a IP-based sensor thati s trying to locate the "right" Cerbo. The problem would probably be solved, as soon as a better solution for issue #768 is found, that really applies the system name as the host name. That (presumably) unique hostname should be enough for avahi to announce a more appropriate name via mDNS.

mpvader, 2023-03-25:
Hi, that sounds like a nice change to do; but also as one that might need some serious testing to make sure it doesn’t break compatibility with for example MFDs.

I’ll keep it in mind.

ps. Is there no serial number for the GX in the mDNS records, and wouldnt it be better for sensors to select the right one on that?

jhofstee, 2026-09-06:
Support for venus-hqnnnnnnn.local has been added in v3.80~37 in addition to venus.local. The hq number can be found in the support status page.

closed 2026-09-06 by jhofstee