victronenergy/venus #1051: Systemcalc crashes cyclically with DVCC enabled

thucar, 2023-03-26:
I'm using dbus-mqtt-device driver to create my devices using MQTT. When I have DVCC enabled the Systemcalc service starts crashing with the following traceback:

```
@40000000641fb9d608c1340c -------- dbus_systemcalc, v2.103 is starting up --------
@40000000641fb9d608dad68c INFO:logger:Loglevel set to INFO
@40000000641fb9d60acd481c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@40000000641fb9d60b0305ec INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
@40000000641fb9d60b4bda74 INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
@40000000641fb9d60eef249c INFO:dbusmonitor:Found: com.victronenergy.battery.socketcan_can0, scanning and storing items
@40000000641fb9d60f2a0ea4 INFO:dbusmonitor:       com.victronenergy.battery.socketcan_can0 has device instance 512
@40000000641fb9d610d7bea4 INFO:dbusmonitor:Found: com.victronenergy.vebus.mqtt_deye_wr, scanning and storing items
@40000000641fb9d6111feb34 INFO:dbusmonitor:       com.victronenergy.vebus.mqtt_deye_wr has device instance 1
@40000000641fb9d615bf3ce4 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@40000000641fb9d636a3443c INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.system
@40000000641fb9d637a01acc INFO:relaystate:Relays found: /dev/gpio/relay_1/value, /dev/gpio/relay_2/value
@40000000641fb9d637c70f74 INFO:buzzercontrol:PWM buzzer found @ frequency: 4000
@40000000641fb9d71e1c1394 INFO:dbus_systemcalc:Battery service initialized to None (setting == nobattery)
@40000000641fb9d72b036bd4 INFO:dbus_systemcalc:Starting mainloop, responding only on events
@40000000641fb9d81e7dafdc exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
@40000000641fb9d81edea044 Traceback (most recent call last):
@40000000641fb9d81ee36ad4   File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/ve_utils.py", line 24, in exit_on_error
@40000000641fb9d81ee38244     return func(*args, **kwargs)
@40000000641fb9d81ee4785c   File "/opt/victronenergy/dbus-systemcalc-py/delegates/dvcc.py", line 959, in _on_timer
@40000000641fb9d81ee48be4     self._multi.firmwareversion is not None and self._multi.firmwareversion < VEBUS_FIRMWARE_REQUIRED))
@40000000641fb9d81ee71454 TypeError: '<' not supported between instances of 'str' and 'int'
```

mpvader, 2023-03-26:
Looks like you are making a com.victronenergy.vebus device with a non numeric value for firmware version.

can I assume its some inverter that is not from victron?

the code works ok with our inverters.

thucar, 2023-03-26:
Indeed, I had added "ALPHA" to the WF version. Thank you for pointing that out to me.

closed 2023-03-26 by thucar