victronenergy/venus #1052: Vebus: Add prefer renewable energy setting

vejpasop, 2023-03-28:
Add a user control to make a solar charger the priority source in providing power to the battery.
This control will tell the Multi to skip the Bulk and Absorption cycle. When enabled, as long as there is sun every day, the unit will stay in float. Otherwise after 12 hours the Multi will go to reduced float.

### Plan:

In the Multi firmware, a settable flag in the eeprom is added, which when enabled makes it:
   1. when not in BOL mode charge to float voltage, and skip bulk and absorption.
   2.  when in BOL mode (canbus battery), charge with the lowest value of the CVL and Float voltage setting.
   3. for a Quattro, this will only work when connected to AC-in-2, since AC-in-1 is where a generator is connected and while that is running you want to charge with full power.
The user control will be made visible in all the UIs:

### Todo:

- [ ] 	VictronConnect (when for example connected via Smart Dongle)
- [ ]	GX Device UI both current one and v2
- [ ]	VRM Dashboard controls
- [ ]	HTML5 UI (only the new one)
- [ ]	VEConfigure

Once a Multi is updated to high enough firmware, all the UIs will show this option automatically

mpvader, 2023-03-28:
hi Kevin, Venus OS is being fixed up to support generators to be connected on AC in 1.

more details here:

https://professional.victronenergy.com/news/detail/178/

kwindrem, 2023-03-28:
The other thing I have heard is that AC input presence (voltage and frequency) are only sensed on the active input and AC input 1. If that is the case then Quattro would not be able to detect mains on AC in 2 and therefore stop the generator. If this is not the case and AC in 2 presence can be detected then all is well.

mpvader, 2023-03-28:
Yes, it can be detected on AC in 2.

vejpasop, 2023-03-31:
User setting `/Dc/0/PreferRenewableEnergy` added to in mk2-dbus v3.52

mpvader, 2023-06-19:
Short update, also @kwindrem , this is progressing quickly now. Above text has been updated, and see here:

https://www.victronenergy.com/live/ve.bus:skip-absorption

vejpasop, 2023-06-20:
More changes are required for generator support.

When a generator is running the override should be active as long as the generator runs to ensure proper charging.
In a systems system with a DMC the DMC can communicate the generator state trough the "transfer switch" input of the DMC.
In a system with a Venus device only we need to communicate generator run state. For this we will need to add an extra dbus path to mk2dbus to communicate the generator state. @Izak will than have to write this path.

jepefe, 2023-06-21:
I added this to the GUI. The battery shows new icons to reflect that the prefer renewable option is enabled, then when a full charge is forced a red strip with icons is added, red is used to maximice the contrast so user can see it from the distance. 

Changes are ready to be added to the todo but first I'll like to get some feedback.

Here is a short video:

https://github.com/victronenergy/venus/assets/6424631/bc62a6e7-b912-45d0-88c6-f1a147ec49c6

jepefe, 2023-06-27:
Handle edge case when 506 firmware settings file is used to restore settings on fw 504 / 505.
Use `/Ac/State/RemoteGeneratorSelected` item to for the logic.

https://github.com/victronenergy/venus/assets/6424631/6c251cb1-d78c-4a13-a87c-e5b7540bac54

When external generator is selected so charge to 100% can't be stopped:

[image]

vejpasop, 2023-06-27:
mk2-dbus v3.57 is on the Venus TODO

vejpasop, 2023-07-27:
@jepefe I noticed an issue with the GUI radio buttons for "Charge the battery to 100%". When you select this option for the first time (after a clean start of the GUI service) the yes / no buttons are selected like 
[image]

After selecting yes the recharge starts correctly. However when you revisit the screen to return to normal operation the "yes" button is selected like 
[image]
Now if you click "yes" the "yes" buttons selection clears and we return to normal operation. The next time we want to start charging the "yes" button is already pre-selected. This looks a bit awkward. See video

https://github.com/victronenergy/venus/assets/10847324/58942304-94e8-42f3-80b1-6d4d66b2e954

izak, 2023-08-22:
@vejpasop, we have sites now that think they have a generator when they don't have one. I think the reason is this:

> * /Ac/State/RemoteGeneratorSelected will be 1 when either of the following is true:
>    * On Quattro while we are connected to AC input 1
>    * When the DMC generator switch input is closed
>    * When '/Ac/Control/RemoteGeneratorSelected' is set to 1

Sites with the grid connected to AC-in-1 of a Quattro now set this path to 1. The GUI then shows that the AC input is a generator.

Probably the simplest solution is to drop that first condition. Or will that lose something important?

kwindrem, 2023-08-22:
There is also a problem with the override of preferred renewable energy if the AC input type is set to generator in the GX device. In this case, the system does not automatically recharge the battery  to 100%. 

The "generator" digital input to say the generator is running sets /Ac/Control/RemoteGeneratorSelected. However this does not happen of the AC input type is manually set to Generator.

Also, are there cases where if the GX device's configuration changes that the /Ac/Control/RemoteGeneratorSelected state isn't updated properly?

My thought is that this all should be handled elsewhere. startstop.py seemed the likely spot however mpvader mentioned that not all users use the generator control logic in the GX device. So maybe in systemcalc?

mpvader, 2023-08-30:
> The GUI then shows that the AC input is a generator.

As discussed, we’re going to remove/revert that out of the gui. @jepefe .

and @vejpasop is going to change mk2dbus to no longer use `Ac/Control/RemoteGeneratorSelected` when the device is a quattro. Since that relates to external transfer switches, which really is for Multis

vejpasop, 2023-08-31:
We should remove the the corresponding generator section of  Solar & Wind Feature video also.

vejpasop, 2023-08-31:
Since the `Ac/State/RemoteGeneratorSelected` was also used by the GUI to determine if we could stop the charge cycle manually we need to add an "overridden" state to  `Dc/0/PreferRenewableEnergy` in case of a running remote generator or a Quattro on AC in 1

vejpasop, 2023-09-04:
In mk2-dbus v3.58  changes 

1. "sustain" is now reported through the `State ` item.

2. `Ac/State/RemoteGeneratorSelected` remains invalid on Quattro's. The current GUI relies on this to determine if the "charge to 100%" can be manually stopped. In case of a generator running or a Quattro on ACin 1 it cannot be stopped. To solve this `Dc/0/PreferRenewableEnergy` now reports "overridden' in this case.

Todo:
- [ ] dbus-systemcalc revert 664f330f9d37dc281fc569aa9500affb4c772da1 @izak 
- [ ] GUI add `SystemStateDescription { value: 0xF4; name: "sustain"; text: qsTr("Sustain")}` @jepefe 
- [ ] GUI use `Dc/0/PreferRenewableEnergy` "overridden" instead of `Ac/State/RemoteGeneratorSelected` @jepefe

jepefe, 2023-09-06:
Changes added in gui: fix_solar_and_wind

closed 2023-09-06 by jhofstee